carla-simulator/carla #3782: Switching Sublevel map does not work

jimmyqwy, 2021-01-08:
Hi Sir, 
I have tried to switch the sublevel map (different layers) through shortcut V / B from Carla client, but it does not work.
Town03_Opt and Town01_Opt maps are tested.
CARLA version - 0.9.11 - download from https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.11.tar.gz

By the way, I have also tested this feature from Editor compiled from source,  it does not work either.
Do I need any special configuration or option/argument for launching this feature?

Thank you!

CARLA version - 0.9.11
OS - Ubuntu 16.04.
Python - 3.7

XGodina, 2021-01-11:
Hi @jimmyqwy,

You can switch between sublevel map with the shortcut "v". When you decide which sublevel you want to Unloaded, press "Shift+b" and b to reload.

When you run the script manual_control.py if you press "h" there is useful info about it.

jimmyqwy, 2021-01-13:
Thank you @XGodina , it works!
Just find some names of sub level do not 100 percent match the list levels from client, but it should be fine.
Thanks again.

closed 2021-01-13 by jimmyqwy

pavanteja295, 2021-07-04:
Hello I'm trying to save images applying the sublevel option to the worlds. However, I do not see the expected changes. I have detailed this here. Any help would be very highly appreciated. 

https://github.com/carla-simulator/carla/issues/3517#issuecomment-873606472.

I'm running the server in synchronous mode though.