carla-simulator/carla #3783: Exporting the scene as a mesh

Ainaz99, 2021-01-08:
Hi,
I'm wondering if there is any way to export the scene surrounding the vehicle as a mesh.

Thank you for your response.

XGodina, 2021-01-11:
Hi @Ainaz99,

There is no way to export automatically the scene around the vehicle, If, it is that what you mean. 
Can you give me more details about it ? Is it your scene ? Is that scene in CARLA ?

Ainaz99, 2021-01-11:
Hi @XGodina,

I would like to be able to extract the mesh of the scene around the car, equivalent to what is scanned by the lidar, in each frame of the simulation. The scene is also in CARLA. 
Thanks.

XGodina, 2021-01-13:
Hi @Ainaz99,

The process I would do is a little manual. First of all, I will select all the meshes around my scene in Unreal editor. After that, click+MR and convert it to static mesh. Now you have one mesh with all that you need.

closed 2021-07-21 by stale[bot]

AlbertHuyb, 2023-04-26:
Hi @XGodina,
I'm quite new to this awesome CARLA simulator. I wonder if there's a better solution to this issue now in 2023. 
Regarding to your method with Unreal, how can I find out which files should be open in Unreal editor? Are there any tutorials about that? 
Thanks.