carla-simulator/carla #3785: How to remove all rider animations for 2 Wheeled vehicles ?

adir42, 2021-01-08:
Hello Carla Team, I would like that all riders of two wheeled vehicles stay static without any animations. How I can remove all the animations of riders ?

XGodina, 2021-01-11:
Hi @Anna492,

I will take a look. As soon as possible I'll tell you something.

Anna492, 2021-02-03:
Hello @XGodina, do you have any solution to this question? Thank you in advance.

Madecu, 2021-02-10:
Hi @Anna492,

In order to enable the two wheels animations, you have to go inside CARLA's UnrealEngine project, find "BP_Base2wheeledNew", select the mesh and change the Anim Class to 'None'

[image]

Anna492, 2021-02-10:
Thank you @Madecu.

closed 2021-02-15 by Anna492