carla-simulator/carla #3786: carla 0.9.11 docker container fails to start

gaoyinfeng, 2021-01-09:
-My Computer:
 Kernel Version: 5.4.0-59-generic
 Operating System: Ubuntu 20.04.1 LTS
 GPU: RTX 2080 Ti
 Nvidia drivers: 455.38 ,with Nvidia-docker2
-Docker:
 Docker version 20.10.2, build 2291f61

I tried to run Carla simulator 0.9.11 as a docker container using this command:
  docker run -it -p 2000-2002:2000-2002 --gpus all carlasim/carla:0.9.11 ./CarlaUE4.sh -carla-server -windowed -ResX=800 -ResY=600 -quality-level=Low --world-port=2000

then i got:
  4.24.3-0+++UE4+Release-4.24 518 0
  Disabling core dumps.
  sh: 1: xdg-user-dir: not found

I checked docker ps and nvidia-smi, clearly the docker container was not created. On the other hand, i used same command to run Carla simulator 0.9.9 as a docker container, and it worked fine. So what should i do to solve this problem? Why this happend?

maro-ros, 2021-01-11:
Have you tried to run:

xhost local:root

before and after that

sudo docker run -e SDL_VIDEODRIVER=x11 -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -p 2000-2002:2000-2002 -it --gpus all carlasim/carla:0.9.11 ./CarlaUE4.sh -opengl

?

gaoyinfeng, 2021-01-11:
@maro-ros 
thanks, it works. And I find the most important command is '-v /tmp/.X11-unix:/tmp/.X11-unix'.

pavanteja295, 2021-01-12:
Hello @maro-ros.  The above solution doesn't work without -opengl option. Could you please propose a solution using -vulkan.

maro-ros, 2021-01-13:
@pavanteja295
Sorry, but I'm also just an end-user :-) - Vulkan also does not work for me btw.

corkyw10, 2021-03-10:
@bernatx could you have a look at this please?

armando-fandango, 2021-03-23:
Any update on this please?

qhaas, 2021-07-07:
>  Could you please propose a solution using -vulkan.

Take a look at [Issue 3800](https://github.com/carla-simulator/carla/issues/3800#issuecomment-768993043)

closed 2023-06-02 by stale[bot]