carla-simulator/carla #3787: Car lights not turning on

maro-ros, 2021-01-11:
Carla 0.9.11 (release) Linux Ubuntu 18.10
================================

I have tried to turn on vehicle lights at night manually via:

```
light_state = carla.VehicleLightState(carla.VehicleLightState.All)

for actor in self.world.get_actors():
    if actor.type_id.startswith("vehicle"):
        actor.set_light_state(light_state)
```

But they are still off. Am I missing something here?

Also trying something on the ego vehicle like
light_state = carla.VehicleLightState.LowBeam | carla.VehicleLightState.Position
vehicle_actor.set_light_state(carla.VehicleLightState(light_state))
is not working.

All other lights are on like street lights or buildings or even bicycles just vehicles stay dark.

Any help is appreciated.

[image]

XGodina, 2021-01-11:
Hi @maro-ros,

I recommend using the environemnt.py to set the lights. With this script, you can play with any light in the scene, including vehicles.

I attach the video in 14:00 explaining how to set the light
https://www.youtube.com/watch?v=ZZaHevsz8W8&t=867s

maro-ros, 2021-01-12:
Hey XGodina,

Thanks for the info. I tried your script. Specifically with the script spawn_npc.py -n 100 --car-lights-on I was able to understand my misconception that "all" cars support lights but this is not the case. After further research I found a list of vehicles in Carla release 0.9.8 that support lights:

> **Bikes**. All of them have a front and back position light.
> **Motorcycles**. Yamaha and Harley Davidson models
> **Cars**. Audi TT, Chevrolet, Dodge (the police car), Etron, Lincoln, Mustang, Tesla 3S, Wolkswagen T2 and the new guests coming to CARLA.

Best regards

XGodina, 2021-01-13:
Hi @maro-ros,

Yes, that is true. If you see the blog of the last release, now we are working to improve our vehicle gallery. We don't have all the vehicles with the lights yet.

closed 2021-07-21 by stale[bot]