carla-simulator/carla #3789: Adding new vehicle to CARLA

kubs0ne, 2021-01-11:
Hello!
I'm new to the CARLA simulator, and I wanted to add my own vehicle to this simulator. I created a model using this tutorial: https://continuebreak.com/articles/how-rig-vehicle-blender-28-ue4/ . Everything works fine, I created blueprints, physics assets and all the necessary components. The cars goes forward and turns as expected. But i'm confused with importing this model to CARLA. How do I do it? I read this tutorial: https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/ , but it doesn't tell where in CARLA there is an option to import blueprints etc. Also, I don't understand the 10th step of the tutorial. How do I open this "vehicle factory"? Any help would be great :).

XGodina, 2021-01-11:
Hi @kubs0ne,
In CARLA there is a folder Carla/vehicles/ where we have all relative of cars. Inside the Blueprint "vehicle factory" there is a array of vehicles where you can put your car.

[image]

kubs0ne, 2021-01-11:
Thank you for quick response. So does it mean I have to open the CarlaUE4.uproj file as a project in UE4?

BAROBA-3d, 2021-01-12:
Yes, you need that project to be able to use the functionality you need. Everything in the docs presumes that you use that project, something to keep it in mind.

parth3153, 2021-02-15:
> Yes, you need that project to be able to use the functionality you need. Everything in the docs presumes that you use that project, something to keep it in mind.

hello, i am struggling with the same thing, but when i open CarlaUE4 Project file it does not open it says 
[image]

and then when i rebuild , it says rebuilding could not be compiled try building from source. btw i have UE 4.24.3.
thank you any help would be appreciated

closed 2021-07-21 by stale[bot]

eslaaam3, 2022-08-18:
Hello,
I'm having a similar issue and I have posted an [issue](https://github.com/carla-simulator/carla/issues/5674) #5674
Can anyone help please?
Thanks!