carla-simulator/carla #3795: Extrinsic Matrix of Lidar to Camera fusion is not constant!

dupply, 2021-01-11:
As mentioned in lidar_to_camera.py, at first lidar points are converted to World frame than to Camera frame and finally to Image frame. 
`            # This (4, 4) matrix transforms the points from lidar space to world space.
            lidar_2_world = lidar.get_transform().get_matrix()
            

            # Transform the points from lidar space to world space.
            world_points = np.dot(lidar_2_world, local_lidar_points)

            # This (4, 4) matrix transforms the points from world to sensor coordinates.
            world_2_camera = np.array(camera.get_transform().get_inverse_matrix())

            # Transform the points from world space to camera space.
            sensor_points = np.dot(world_2_camera, world_points)`

Usually after the extrinsic matrix is constant because relative postion of the sensors are not changing but extrinsic matrix for projection of lidar point to camera frame ie `np.dot(world_2_camera, lidar_2_world)` is not constant.

I am asking this because I am writng a C++ ROS code for projection of lidar point cloud to image frame and I am subscribing only lidar and camera rostopics

closed 2021-01-12 by dupply

makaveli10, 2022-05-30:
@dupply did you get the lidar_2_cam matrix finally?

jiheddachraoui, 2023-04-14:
@dupply  did you find an answer

zhiranyan, 2023-08-01:
Hi, i am also curious about how to get extrinsic matrix between lidar and camera. Is there any way? Thanks a lot!