carla-simulator/carla #3799: [Feature Request] HDF5 or LMDB output

qhaas, 2021-01-13:
It would be useful if the CARLA client could output raw imagery data / metadata to a single and/or [memory-mapped file](https://en.wikipedia.org/wiki/Memory-mapped_file) (e.g. [lmdb](https://en.wikipedia.org/wiki/Lightning_Memory-Mapped_Database) or [hd5f](https://en.wikipedia.org/wiki/Hierarchical_Data_Format)).  We are experiencing IO limitations on large runs, even with several local SSDs in RAID0, and we suspect that it is likely due to the generation large amounts of tiny, individual image files per second (and the overhead associated with opening/closing their FDs, writing disparate / non-sequential locations, etc).

corkyw10, 2021-03-11:
@Axel1092 could you follow up on this please?

Axel1092, 2021-03-16:
Hi @qhaas, I don't think I am really understanding the issue. You would like to save to disk several images at the same time to speed up disk usage?

qhaas, 2021-03-16:
We are running CARLA simulations in synchronous mode on high performance computing (HPC) systems (e.g. the [Summit supercomputer](https://github.com/carla-simulator/carla/issues/1567#issuecomment-700790499)) and some nvidia DGX systems.  Our DGXs and HPC systems typically use [clustered file systems](https://en.wikipedia.org/wiki/Clustered_file_system) (e.g. [lustre](https://en.wikipedia.org/wiki/Lustre_(file_system))).  Clustered file systems are not great at dealing with [large numbers of small files in a single directory](https://hpcf.umbc.edu/general-productivity/lustre-best-practices/) (e.g. synchronous CARLA runs lasting several hours at 30FPS saving imagery to PNG files that are < 2MB).  HPC runs are an edge case for CARLA, but scaling to 'big iron' systems is likely necessary to increase test coverage of autonomous systems.

We are exploring implementing this functionality in CARLA ourselves, since HPC is likely a low priority for the upstream CARLA development team.  Upon examination of the [CARLA code](https://github.com/carla-simulator/carla/blob/0.9.11/LibCarla/source/carla/image/ImageIO.h#L23), it uses [BOOST GIL](https://www.boost.org/doc/libs/1_68_0/libs/gil/doc/html/index.html) to handle 'high-level' imagery I/O, which doesn't seem to support common HPC data file formats like HDF5 or LMDB.  I suspect we will first need to write a BOOST GIL plug-in to save images in one of said formats, then implement (throughout the CARLA PythonAPI/C++ API) a way to specify a frame 'offset' so the client can know where to save the new frames / metadata in an [hdf5 image file](https://portal.hdfgroup.org/pages/viewpage.action?pageId=48808278).

Axel1092, 2021-03-16:
@qhaas indeed carla does not provide any support for these systems. You can aces the raw data of the images with `carla.Image.raw_data` and store it yourself as it works best for you.

closed 2023-06-02 by stale[bot]