carla-simulator/carla #3801: V0.9.11 Some 2 wheeled vehicle cannot get bounding box.

Chuck-Yu, 2021-01-14:
[image]

Platform: Ubuntu 16.04
CARLA version: 0.9.11
Script: Python/example/client_bounding_boxes.py

xinshuoweng, 2021-01-23:
Same here. This is only happening to some cyclists and motorcycles.

XGodina, 2021-01-27:
Hi, I will check. Thanks for your report.

xinshuoweng, 2021-01-27:
From what I observed, it is not that the bounding boxes for some 2-wheel vehicles are missing. Instead, the boxes are in the wrong position because some boxes appear in places without any object in my visualization. Also, this situation does not happen to all instances of 2-wheel vehicles, some are still good with correct boxes. 

For those instances with wrong boxes, they seem to be easily identified. Those instances all have a width of 0 (vehicle.bounding_box.extent.y). So my current solution is simply to remove vehicles with a width of 0 after spawning several hundreds of vehicles, which removes about half of the 2-wheel vehicles. The rest are working perfectly.

jamesgunnfiveai, 2021-02-02:
I believe this is related to https://github.com/carla-simulator/carla/issues/3670 as described by @xinshuoweng. The problematic vehicles (0 y-extent) are only ever two-wheelers (as described in 3670) and end up with incorrect bounding box locations.

Karthikeyanc2, 2021-05-03:
I am also facing this problem.

closed 2021-07-21 by stale[bot]

mawi42, 2021-09-15:
With Carla 0.9.12 I still see this problem

yuvalHG, 2021-12-14:
@XGodina , is there any new with this issue?

XHwind, 2022-03-31:
Faced the same problem with 2-wheel vehicles. Are there any updates?

DerrickXuNu, 2022-03-31:
Why close this issue? it is not solved yet

shuishida, 2023-07-07:
I have had similar problems with the bounding boxes, and this is my wrapper function to correct for any misbehaving bounding boxes.

`
def get_bounding_box(actor: Actor, min_extent=0.5):
    """
    Some actors like motorbikes have a zero width bounding box. This is a fix for these issues.
    """
    # Fixing bug related to Carla 9.11 onwards where the bounding box location is wrongly registered for some 2-wheeled vehicles
    # https://github.com/carla-simulator/carla/issues/3801
    buggy_bbox = (bbox.extent.x * bbox.extent.y * bbox.extent.z == 0)
    if buggy_bbox:
        # print(f"Buggy bounding box found for {actor}")
        bbox.location = carla.Location(0, 0, max(bbox.extent.z, min_extent))
    # Fixing bug related to Carla 9.11 onwards where some bounding boxes have 0 extent
    # https://github.com/carla-simulator/carla/issues/3670
    if bbox.extent.x < min_extent:
        bbox.extent.x = min_extent
    if bbox.extent.y < min_extent:
        bbox.extent.y = min_extent
    if bbox.extent.z < min_extent:
        bbox.extent.z = min_extent

    return bbox
`

shuishida, 2023-07-07:
Yes exactly - this is because the bounding box location is not set properly. I’m just setting it to x=0 and y=0 here. This seems to work since the bounding box location is defined relative to the actor coordinate frame.

KavuSlayer, 2023-07-13:
Through trial and error found a good enough solution.
The thing is, that the location and extent of some models are just seem to be switched places. So we just have to check for bug, as @shuishida suggested and then switch the data:

`
def get_bounding_box(actor: Actor, min_extent=0.5):
    """
    Some actors like motorbikes have a zero width bounding box. This is a fix to this issue.
    """
    if not hasattr(actor, "bounding_box"):
        # print(f"Bounding box for type {actor.type_id} not implemented. Returning default.")
        return carla.BoundingBox(carla.Location(0, 0, min_extent), 
                                 carla.Vector3D(x=min_extent, y=min_extent, z=min_extent))
    
    bbox = actor.bounding_box
    # Fixing bugs related to Carla 9.11 onwards where the bounding box location and extent is wrongly registered for some 2-wheeled vehicles
    # https://github.com/carla-simulator/carla/issues/3801
    # https://github.com/carla-simulator/carla/issues/3670
    buggy_bbox = (bbox.extent.x * bbox.extent.y * bbox.extent.z == 0)
    if buggy_bbox:
        # print(f"Buggy bounding box found for {actor}")
        loc = carla.Location(bbox.extent)
        bbox.extent = carla.Vector3D(bbox.location)
        bbox.location = loc
    return bbox
`

2mey10, 2024-06-28:
Hey, if anyone still sees this thread, I have created a fix for my carla-KITTI framework.

`bicycle_fix_dict = {
        "vehicle.vespa.zx125": {
            "ext": [0.8, 0.45],
            "sensor_refpoint": [0.6,0.45],
        },
        "vehicle.bh.crossbike": {
            "ext": [0.8, 0.45],
            "sensor_refpoint": [0.6, 0.45],
        },
        "vehicle.gazelle.omafiets": {
            "ext": [0.75, 0.3],
            "sensor_refpoint": [0.9, 0.13],  
        },
        "vehicle.yamaha.yzf": {
            "ext": [0.9, 0.45],
            "sensor_refpoint": [0.5, 0.4],
        },
        "vehicle.kawasaki.ninja": {
            "ext": [0.9, 0.35],
            "sensor_refpoint": [0.65,0.4],
        },
        "vehicle.harley-davidson.low_rider": {
            "ext": [0.9, 0.35],
            "sensor_refpoint": [0.65,0.4],
        },
        "vehicle.diamondback.century": {
            "ext": [0.7, 0.25],
            "sensor_refpoint": [0.7,0.2],
        },
        }

    if obj_type == "Bicycle" or obj_type == "Motorcycle":
        log.debug("new ext: %s", ext)
        if agent.type_id in bicycle_fix_dict:
            ext.x = bicycle_fix_dict[agent.type_id]["ext"][0]
            ext.y = bicycle_fix_dict[agent.type_id]["ext"][1]

            sensor_refpoint[0] += bicycle_fix_dict[agent.type_id]["sensor_refpoint"][0]
            sensor_refpoint[1] += bicycle_fix_dict[agent.type_id]["sensor_refpoint"][1]`

You can ignore the sensor_refpoint and instead manipulate the location in the same way.
I hopy this fix can help anyone that has the same problem.
This works in carla 0.9.14 :)