carla-simulator/carla #3804: Can Carla configure the angle between each laser in the lidar?

KangChou, 2021-01-17:


MisterioRemo, 2021-01-18:
Hello!
I think it's impossible now. Lidar has a constant step between rays.
But you can create an attribute for the lidar class (for example _angles_array_) and pass it as a string. Then in ARayCastSemanticLidar modify `CreateLasers` function (or create your own) where you parse the _angles_array_ into a list of floats and put each element into _LaserAngles_ array in rifght order.

KangChou, 2021-01-23:
Thank you very much for your answer!
About the above questions, can you write a script for me?
Because I found that many R &amp; D personnel have this requirement. I hope you can provide a reference for an example. Thank you very much.
At the same time, I also hope to contribute to the Carla platform.

MisterioRemo, 2021-01-26:
Ok, I try doing it.
I have my own lidar class, so I've never changed RayCastSemanticLidar before. A code below is just a guess, I didn't test it.

1. Add attribbute to the lidar class.
[Create a sensor](https://carla.readthedocs.io/en/latest/tuto_D_create_sensor/#1-sensor-actor) tutorial describes how to do this for usual case. For RayCastSemanticLidar it's almost the same.

   In _LidarDescription.h_ add new LaserAngles varible.
   ```cpp
   USTRUCT()
   struct CARLA_API FLidarDescription
   {
     GENERATED_BODY()

     // code

     UPROPERTY(EditAnywhere)
     FString LaserAngles = "10.0, 5.0, 2.0, 0, -2.0, -5.0, -10.0";
   };
   ```
   In _ActorBlueprintFunctionLibrary.cpp_ modify `MakeLidarDefinition` and `SetLidar` functions.
   ```cpp
   
   void UActorBlueprintFunctionLibrary::MakeLidarDefinition(
       const FString &Id,
       bool &Success,
       FActorDefinition &Definition)
   {
     // function code
   
     FActorVariation Angles;
     Angles.Id = TEXT("laser_angles");
     Angles.Type = EActorAttributeType::String;
     Angles.RecommendedValues = { TEXT("10.0, 5.0, 2.0, 0, -2.0, -5.0, -10.0") };
   
     // add Angles to Definition.Variations
     if (Id == "ray_cast") {
       Definition.Variations.Append(
           {Channels, Range, PointsPerSecond, Frequency, UpperFOV, LowerFOV,
               AtmospAttenRate, DropOffGenRate, DropOffIntensityLimit,
               DropOffAtZeroIntensity, StdDevLidar, Angles });
     }
     else if (Id == "ray_cast_semantic") {
       Definition.Variations.Append(
           {Channels, Range, PointsPerSecond, Frequency, UpperFOV, LowerFOV, Angles });
     }
     else {
       DEBUG_ASSERT(false);
     }
   
     Success = CheckActorDefinition(Definition);
   }
   
   void UActorBlueprintFunctionLibrary::SetLidar(
       const FActorDescription &Description,
       FLidarDescription &Lidar)
   {
     // function code
   
     Lidar.LaserAngles =
         RetrieveActorAttributeToString("laser_angles", Description.Variations, Lidar.   LaserAngles);
   }
   ```

2. Parse LaserAngles value.
   In _RayCastSemanticLidar.cpp_ change `Set(const FLidarDescription &LidarDescription)` and `CreateLasers()`.
   ```cpp
   void ARayCastSemanticLidar::Set(const FLidarDescription &LidarDescription)
   {
     Description = LidarDescription;
     CreateLasers();
     SemanticLidarData = FSemanticLidarData(Description.Channels);
     PointsPerChannel.resize(Description.Channels);
   }
   
   void ARayCastSemanticLidar::CreateLasers()
   {
     TArray<FString> TempAnglesArray;
     Description.LaserAngles.ParseIntoArray(TempAnglesArray, TEXT(","), true);
     if (Description.Channels != TempAnglesArray.Num()) {
       UE_LOG(LogCarla, Warning, TEXT("Channels is set to %d"), TempAnglesArray.Num());
       Description.Channels = static_cast<uint32>(TempAnglesArray.Num());
     }
     check(Description.Channels > 0u);
     LaserAngles.Reset(Description.Channels);
   
     for (auto Angle : TempAnglesArray)
       LaserAngles.Emplace(FCString::Atof(*Angle));
     // sort LaserAngles if needed, because the order of the items in LaserAngles is important
   }
   ```

3. Rebuild carla.
4. That's all, I think. Now in python for lidar blueprint we can do something like this:
   `bp_lidar.set_attribute('laser_angles', '-5, -1, 0, 2, 10')`.

KangChou, 2021-01-27:
Great. Thank you for your generous contribution.
If it's convenient for you, can you provide Python code again?
This is because I will use it for image processing, and python will easily access my algorithm program.

I will be very grateful to you!

@MisterioRemo @rowanmcallister

MisterioRemo, 2021-01-27:
@KangChou Sorry, but I don't understand what exactly you want me to write.
We need to change class code in c++ to be able to set lidar angles from string attribute (list of floats).
In python we can only set attribute value.