carla-simulator/carla #3809: Sensors ignore sensor_tick attribute

tobu1, 2021-01-19:
Hi,

the sensors camera.dvs, lidar.ray_cast, lidar.ray_cast_semantic, other.gnss, other.imu and other.obstacle ignore the sensor_tick attribute. That means they send data for every frame.  The issue for me is that I need exactly 25 datasets per seconds from the lidar.ray_cast_semantic sensor. 

I am using Carla 0.9.11 on Windows 10.

MisterioRemo, 2021-01-22:
I have the same problem. It seems like sensors were changed a bit in the new version of Carla.
Sensor_tick attribute changes tick interval for `Tick` function. But now data from the sensor is sent in `PostPhysTick` which is called every world tick.
So attribute works, it just affects wrong function.

goksanisil23, 2021-01-26:
Hi, I have similar problem in 0.9.10.1:

I'm trying to get different sensor update rates via ros-bridge, by configuring the sensors.json file. However, I cannot see sensor_tick parameter affecting the outcome. So far, I've debugged it including the individual sensor source code inside Unreal. 

I can see that inside ASensor::Set function, SetActorTickInterval function sends correct sensor_tick parameter I set from json file.

However, when I print out the DeltaTime and time passed since actor creation inside individual sensor implementations (GnssSensor.cpp, RayCastLidar.cpp, etc), I always see time difference as fixed_delta_seconds parameter I set from launch file. Therefore all my sensor topic publications have the same rate (note: running in async mode)

Summary: 
synchronous_mode : false
fixed_delta_seconds: 0.01
sensor.json: lidar-> sensor_tick: 0.1
--------------------------
outcome: lidar topic at 100Hz

mawi42, 2021-01-26:
Hi,

for 0.9.10.4 when set to 100Hz tick rate and image sensor 25Hz I received frames also with 100Hz. Fot sensors Lidar and GNSS 10Hz was OK.
Now 0.9.11 seems different images frame rate as expected 25Hz but GNSS Lidar 100Hz.
For me this a very important feature, would be nice if you could fix it please.

Thanks 
mawi

MisterioRemo, 2021-01-27:
@mawi42 
> for 0.9.10.4 when set to 100Hz tick rate and image sensor 25Hz I received frames also with 100Hz. Fot sensors Lidar and GNSS 10Hz was OK.

It's known issue for all camera sensors in 0.9.10. And it was fixed for 0.9.11 ([pull request](https://github.com/carla-simulator/carla/pull/3399)).

mawi42, 2021-01-27:
> @mawi42
> 
> > for 0.9.10.4 when set to 100Hz tick rate and image sensor 25Hz I received frames also with 100Hz. Fot sensors Lidar and GNSS 10Hz was OK.
> 
> It's known issue for all camera sensors in 0.9.10. And it was fixed for 0.9.11 ([pull request](https://github.com/carla-simulator/carla/pull/3399)).

Hi MisterioRemo, 
thank you for your support and sorry maybe I wrote not clearly. Yes I recognized that camera is working well now. But as you mentioned and also from other users, other ticks now do not work as expected.

BR mawi

MisterioRemo, 2021-01-27:
@mawi42 Oh no! It's me who should apologize. Express myself in English is a bit difficult, so my sentences become shorter and less informative. Sorry.
I actually meant that if you need to find solution as fast as possible and don't have time to wait until carla's team fix this bug, then you can return to 0.9.10 version and temporary apply change from pull requst. So all sensors will work fine.

corkyw10, 2021-03-11:
@DSantosO Could you have a look at this please?

DSantosO, 2021-03-12:
Hello everyone,

This issue was fixed in PR #3818 so the sensor_tick attribute should work for all sensors. You can check it if you compile from source or in our [overnight build](http://carla-releases.s3.amazonaws.com/Linux/Dev/CARLA_Latest.tar.gz). Thank you for the report.

closed 2021-03-16 by DSantosO

Miknow78, 2022-01-05:
> Hello everyone,
> 
> This issue was fixed in PR #3818 so the sensor_tick attribute should work for all sensors. You can check it if you compile from source or in our [overnight build](http://carla-releases.s3.amazonaws.com/Linux/Dev/CARLA_Latest.tar.gz). Thank you for the report.

> `PostPhysTick`

Hello ! Did you already fix the problem? 
I need to reduce the frequency of a collision sensor that doesn't have the sensor_tick attribute.