carla-simulator/carla #3810: Understanding vectors in CARLA

NishanthARao, 2021-01-20:
Hello all. I've been using CARLA 0.9.10 for a while now, and I am still not able to understand (intuitively) what a vector is in CARLA.

For eg., consider the ```GetForwardVector()``` function. After going through it's source code, I understood that the point is transformed by rotating it and translating it by the respective Euler angles and transformation matrices. The function then returns the transformed point. However, a vector would theoretically require two points for it's definition right? (one to define it's initial position, and the other to define the end point). So my question is, How is this vector defined in this case, as the function returns only the transformed point? Is it assumed that the initial point is the center of mass of the vehicle?

Any insights/ hints would immensely help me.  Thank you so much.

corkyw10, 2021-03-11:
@doterop could you have a look at this please?

doterop, 2021-03-23:
Hi @NishanthARao ,

The forward, up and right vectors are calculated based on the transform's rotation.
You don't need two points to calculate them.

closed 2021-03-23 by doterop

NishanthARao, 2021-03-24:
@doterop Thanks for the answer. However, I think there is a misunderstanding. Of course, as you mentioned, the forward, up, and right vectors are calculated based on the transform's rotation using coordinate transformation matrices, and this is also what I mentioned in my question.

The doubt arrives here. For a vector to be defined "mathematically" you need two points. The direction of a vector then is from the initial point (usually chosen as the origin, but need not be) to the final point. In CARLA, the final point alone is the one that is getting transformed via rotation. My question is where is the initial point for that forward, up, and right vector? Is it assumed that the initial point is on the car's center of mass?

Thanks

gunki-geek, 2022-03-14:
It does not return a point but the cartesian coordinates of a vector (it is not attached to a specific start and end point but it is up to you to apply it to whatever start point you want to get its corresponding end point via the translation defined by the vector) 
For example lets say you apply the function GetForwardVector() and you get as output (2,3,0).
then you can apply it to translate a point at coordinate (0,0,0) to (2,3,0) or a point at (1,2,3) to (3,5,3) etc ...