carla-simulator/carla #3811: LibCarla.client.rss build error

Vela-zz, 2021-01-20:
Error Log:

> Starting >>> ad_physics
> --- stderr: ad_physics                          
> CMake Error at /home/unamattina/carla/Build/boost-1.72.0-c8-install/lib/cmake/Boost-1.72.0/BoostConfig.cmake:120 (find_package):
>   Found package configuration file:
> 
>     /home/unamattina/carla/Build/boost-1.72.0-c8-install/lib/cmake/boost_python-1.72.0/boost_python-config.cmake
> 
>   but it set boost_python_FOUND to FALSE so package "boost_python" is
>   considered to be NOT FOUND.  Reason given by package:
> 
>   No suitable build variant has been found.
> 
>   The following variants have been tried and rejected:
> 
>   * libboost_python36.so.1.72.0 (3.6, Boost_PYTHON_VERSION=2.7)
> 
>   * libboost_python36.a (3.6, Boost_PYTHON_VERSION=2.7)
> 
> Call Stack (most recent call first):
>   /home/unamattina/carla/Build/boost-1.72.0-c8-install/lib/cmake/Boost-1.72.0/BoostConfig.cmake:185 (boost_find_component)
>   /usr/local/share/cmake-3.17/Modules/FindBoost.cmake:444 (find_package)
>   /home/unamattina/carla/Build/ad-rss-4.0.0/src/map/cmake/python-binding.cmake:55 (find_package)
>   python/CMakeLists.txt:24 (find_python_binding_packages)
> 
> 
> ---
> Failed   <<< ad_physics [0.58s, exited with code 1]
> 
> Summary: 3 packages finished [32.8s]
>   1 package failed: ad_physics
>   4 packages had stderr output: PROJ4 ad_map_opendrive_reader ad_physics spdlog
>   3 packages not processed
> Util/BuildTools/Linux.mk:146: recipe for target 'ad-rss' failed
> make: *** [ad-rss] Error 1
> 

on Ubuntu 18.04 
python3.6.9 
no conda or venv virtual environments

corkyw10, 2021-03-11:
@fpasch could you follow up on this please?

berndgassmann, 2021-03-11:
Looks as if python2 and 3 are being mixed up in that build?
Boost is compiled for 3.6, but ad_physics is intended to be built using python 2.7?
All has to be compiled for the same python version....

Vela-zz, 2021-03-11:
> Looks as if python2 and 3 are being mixed up in that build?
> Boost is compiled for 3.6, but ad_physics is intended to be built using python 2.7?
> All has to be compiled for the same python version....

Thank you, I found that ad_rss lib was difficult to be built in virtual environment, because the build tool cannot locate right directory of python in venv. So I deactivate all the virtual environment and set py3.6 as default python on Linux without clean built file. Maybe this is the why this error log comes out, thanks for your help

berndgassmann, 2021-06-16:
Issue solved. Closing.

closed 2021-06-16 by berndgassmann