carla-simulator/carla #3812: Specific Python Version for CARLA-0.9.10

iiJoeyii, 2021-01-20:
Hi! I was wondering if there needs to be a specific python version for CARLA, specifically version 0.9.10.
I know that for python 3.9, the PythonAPI codes work well but when I try using 3.8 it introduces a "ModuleNotFound" error. Note the image below.

This is using python version 3.8
[image]

I would like to know if I can use python 3.8 with this version of CARLA and how I can go about it doing so.
Thanks!

MisterioRemo, 2021-01-22:
Hello.
Did you build egg file for python 3.8?

iiJoeyii, 2021-01-22:
Hi, I tried doing so by using 'make PythonAPI' command in the windows build instructions, being sure to be in the x86 Native Tool Command Prompt for VS and also tried using the 'python setup.py install' command in the normal command prompt.

In both cases, I made sure to be using python 3.8. the egg file still showcases for 3.9, even when I uninstall 3.9 from my system.

Maybe I am doing something wrong?

MisterioRemo, 2021-01-22:
> Hi , I tried doing so by using 'make PythonAPI' command in the windows build instructions, being sure to be in the x86 Native Tool Command Prompt for VS  

Hmm, i think now we should be using x64 command prompt. At least, lastest [doc](https://carla.readthedocs.io/en/latest/build_windows/#visual-studio-2017) say so.

When I want to specify python version I use --python-version argument. In your case: `make PythonAPI ARGS="--python-version=3.8"`. I work with python on Ubuntu, but I guess this command works for Windows too.

> In both cases, I made sure to be using python 3.8. the egg file still showcases for 3.9, even when I uninstall 3.9 from my system.

Well, it's strange if you already uninstalled 3.9 version. As far as I know, `make PythonAPI ` uses default python version.
You can check in terminal which python version sets by default. Also, maybe environment variable for python didn't change after you deleted 3.9?

> I tried these builds again, also trying to do another method of building, 'pip install carla/' in the PythonAPI folder and I see in all cases there are errors that occur. The following is the latest message:

Unfortunately, I don't know what it means =(

iiJoeyii, 2021-01-22:
Thank you for the feedback! 

I tried rebuilding again, this time deleting the 'carla/Build/boost-install' directories, and then `make PythonAPI ARGS="--python-version=3.8"`.

This allowed me to finally be able to successfully obtain the egg file for python 3.8 with no errors with regard to building. And `import carla` works just fine.

However, I end up getting the following error:
[image]

Is there a way to work around this error?

MisterioRemo, 2021-01-23:
Look at the [3083](https://github.com/carla-simulator/carla/issues/3083) issue, maybe their solution helps you too.

iiJoeyii, 2021-01-25:
@MisterioRemo Thank you for your help! I'll see if I can get around the error and post the solution if I can. Thank you for the resource as well!

iiJoeyii, 2021-02-07:
It has been a while but I got it working!

There were a few things I tried to make this work so forgive me if this is a jumble of methods, but who knows, this might help someone hopefully.

A few things to note. For my project,  I use the "WindowsNoEditor" version of Carla, the Quick Installation. But I used the "Windows Build" version of Carla to build the specific *.egg file I needed for a python version, in my case Python 3.8.

Method 1:
I made sure to install a specific version of a library solver, using Axel1092's message in issue #3083 
`pip install -Iv setuptools==47.3.1`

I then rebuilt the PythonAPI for carla in the "Windows Build" version using this
> I tried rebuilding again, this time deleting the 'carla/Build/boost-install' directories, and then `make PythonAPI ARGS="--python-version=3.8"`.

I then took the *.egg file that should be in the /PythonAPI/carla/dist directory in the "Windows Build" version and placed in the /PythonAPI/carla/dist directory in the "WindowsNoEditor" version. Carla python scripts work perfectly for me.

Method 2: (This one might be a coincidence)
Still in the "Windows Build" version, there were times before this when the `make PythonAPI` command would result in errors and would not create the *.egg file. In windows, a message would pop up at the end saying the the build was successful, but most times this is not true. If you do not see a *.egg file in the /PythonAPI/carla/dist directory, the build was not successful.

I ended up trying to follow this YouTube [tutorial](https://www.youtube.com/watch?v=AyugrrVJmhA). I followed it after they installed "Windows Build" version of carla at around the time 5:20. They use mingw and uses the command `mingw32-make rebuild` before building python. After following the tutorial, I found that the `make PythonAPI` command works successfully and being sure to do the first method I mentioned, Carla python scripts should work just fine.

Hope this helps someone! I'll keep this issue open for a few more days, if someone else has any problems or questions.
Thanks!

closed 2021-02-15 by iiJoeyii

Mohammed012345, 2021-07-07:
Hey @iiJoeyii 
This thread has been closed for a while, so you might not notice my comment. But if you do, I request some help from you.
I have tried to build the .egg file for python 3.8, but while running `make PythonAPI ARGS="--python-version=3.8"` I keep getting lots of boost errors and can't successfully build the file. It looks like it will take a lot of my time to fix all the errors, so I thought if you can share your .egg file, I can move onto the next part of my project sooner as my deadlines keeps getting closer.

It would be a great help if you can share the egg file.
Thanks!

iiJoeyii, 2021-07-07:
Hi @Mohammed012345, I can try help!

Before you try my egg file try removing all the boost files from the carla/Build directory and rerun `make PythonAPI ARGS="--python-version=3.8"` and see if that works for you.

In any case, I uploaded my egg file to a [miscellaneous repository](https://github.com/iiJoeyii/miscellaneous) I have.

I hope this helps!

Mohammed012345, 2021-07-08:
Hi @iiJoeyii ,
First of all, thanks for the quick response and the egg file :)

I will try removing the boost files and execute the command if I need to in the future. One final question I had was if this egg file would also work with CARLA 0.9.10 version or if not then how to build for a specific version. As far as I've seen, the build always generates egg for the latest version.

Thanks again!

iiJoeyii, 2021-07-08:
Hi @Mohammed012345,
No worries!

You can try to see if it would work with 0.9.10. I should have mentioned that this egg file is for the 0.9.11 version, that's my mistake. Building the egg file should be dependent on what version you have installed. So, if you have 0.9.10, the egg file will build for 0.9.10.

TrinhTuanHung2021, 2021-08-13:
> Hi @Mohammed012345, I can try help!
>
> Before you try my egg file try removing all the boost files from the carla/Build directory and rerun `make PythonAPI ARGS="--python-version=3.8"` and see if that [...]

Could you tell me how to make your egg? Because when I run the command "make PythonAPI" on  Native tool Command, it often has an error. Or could you show me the video on youtube show the process. Thanks

iiJoeyii, 2021-08-13:
> It has been a while but I got it working!
> 
> There were a few things I tried to make this work so forgive me if this is a jumble of methods, but who knows, this might help someone hopefully.
> 
> A few things to note. For my project, I use the "WindowsNoEditor" version of Carla, the Quick Installation. But I used the "Windows Build" version of Carla to build the specific *.egg file I needed for a python version, in my case Python 3.8.
> 
> Method 1:
> I made sure to install a specific version of a library solver, using Axel1092's message in issue #3083
> `pip install -Iv setuptools==47.3.1`
> 
> I then rebuilt the PythonAPI for carla in the "Windows Build" version using this
> 
> > I tried rebuilding again, this time deleting the 'carla/Build/boost-install' directories, and then `make PythonAPI ARGS="--python-version=3.8"`.
> 
> I then took the *.egg file that should be in the /PythonAPI/carla/dist directory in the "Windows Build" version and placed in the /PythonAPI/carla/dist directory in the "WindowsNoEditor" version. Carla python scripts work perfectly for me.
> 
> Method 2: (This one might be a coincidence)
> Still in the "Windows Build" version, there were times before this when the `make PythonAPI` command would result in errors and would not create the *.egg file. In windows, a message would pop up at the end saying the the build was successful, but most times this is not true. If you do not see a *.egg file in the /PythonAPI/carla/dist directory, the build was not successful.
> 
> I ended up trying to follow this YouTube [tutorial](https://www.youtube.com/watch?v=AyugrrVJmhA). I followed it after they installed "Windows Build" version of carla at around the time 5:20. They use mingw and use the command `mingw32-make rebuild` before building python. After following the tutorial, I found that the `make PythonAPI` command works successfully and being sure to do the first method I mentioned, Carla python scripts should work just fine.
> 
> Hope this helps someone! I'll keep this issue open for a few more days, if someone else has any problems or questions.
> Thanks!

Hi @TrinhTuanHung2021, try this for the version of Carla you have and see if that will work.