carla-simulator/carla #3814: Not sure why I can't retrieve event camera data

ivannson, 2021-01-20:
I'm trying to get event camera data following the `manual_control.py` example, however it doesn't seem to be working and I'm not quite sure why it's not working. It's as if the callback function is not bein called at all. I'm trying to record 100 images with a `for` loop, however it just quits after 2-3 iterations, without doing anything and skipping over the `finally` statement which I find confusing. 

Here's the script that I'm trying to run:
```python 
import glob
import os
import sys
import traceback
from sensor_processing.event_cam.event_cam_processing import *

python_api_dir = 'carla/PythonAPI'
try:
    sys.path.append(glob.glob(f'{python_api_dir}/carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])

    print(glob.glob(f'{python_api_dir}/carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    print('Failed')
    pass
from carla import ColorConverter as cc

import random
import time

def main():
    actor_list = []
    client = carla.Client('localhost', 2000)
    client.set_timeout(2.0)
    world = client.get_world()
    blueprint_library = world.get_blueprint_library()
    bp = random.choice(blueprint_library.filter('vehicle'))
    car_transform = world.get_map().get_spawn_points()[156]
    waypoint = world.get_map().get_waypoint(car_transform.location)
    vehicle = world.spawn_actor(bp, car_transform)
    actor_list.append(vehicle)
    print('created %s' % vehicle.type_id)
    #vehicle.set_autopilot(True)
    camera_bp = blueprint_library.find('sensor.camera.dvs')
    camera_transform = carla.Transform(carla.Location(x=1.5, z=2.4))
    camera = world.spawn_actor(camera_bp, camera_transform, attach_to=vehicle)
    print(camera)
    actor_list.append(camera)
    print('created %s' % camera.type_id)
    def nah():
        return None
    try:
        i = 0
        for i in range(100):
            camera.listen(lambda image: event_cam_callback(image, just_events = False, image_events = False, save = True, save_dir = '\\data\\event_cam'))

            print(i)
            waypoint = waypoint.next(1)[0]
            vehicle.set_transform(waypoint.transform)
    except Exception as e:
        print(e)
        traceback.print_exc()
    finally:
        print('destroying actors')
        camera.destroy()
        client.apply_batch([carla.command.DestroyActor(x) for x in actor_list])
        print('done.')

if __name__ == '__main__':

    main()
```

And here's the callback function:

```python 
def event_cam_callback(raw_events, just_events = True, image_events = False, save = False, save_dir = None):
    # Example of converting the raw_data from a carla.DVSEventArray
        # sensor into a NumPy array and using it as an image
    print('event cam callback')
    #raw_events.convert(cc.Raw)
    dvs_events = np.frombuffer(raw_events.raw_data, dtype=np.dtype([
        ('x', np.uint16), ('y', np.uint16), ('t', np.int64), ('pol', np.bool)]))
    
    print(len(dvs_events))
    if just_events and save:
        saved = np.save(Path(save_dir+'/raw/raw_{}.npy'.format(raw_events.frame)), dvs_events)
        print(saved)
    if image_events:
        dvs_img = np.zeros((raw_events.height, raw_events.width, 3), dtype=np.uint8)
        print(len(dvs_img))
        # Blue is positive, red is negative
        dvs_img[dvs_events[:]['y'], dvs_events[:]['x'], dvs_events[:]['pol'] * 2] = 255
        if save:
            saved = np.save(Path(save_dir+'/img/img_{}.npy'.format(raw_events.frame)), dvs_img)
            print(saved)

    if just_events and image_events:
        return dvs_events, dvs_img
    
    if just_events and not image_events:
        return dvs_events
    
    if image_events and not just_events:
        return dvs_img

def do_nothing():
    pass

```

When I use `do_nothing` it fails as well, however if I get rid of `camera.listen` altogether it's fine, so I must be doing something wrong when trying to retrieve the event data.

MisterioRemo, 2021-01-22:
Your script stops before any data from the camera is received, I suppose. Use `time.sleep` or something like that.

```py
camera.listen(lambda image: event_cam_callback(image, just_events = False, image_events = False, save = True, save_dir = '\\data\\event_cam'))
try:
        i = 0
        for i in range(100):
            print(i)
            waypoint = waypoint.next(1)[0]
            vehicle.set_transform(waypoint.transform)
       time.sleep(5.0)
```

Also, take some pause between `set_transform`. You teleport the vehicle too fast so the sensor in example above will capture only the last frame.

closed 2021-03-11 by germanros1987

BehnazFayazi, 2023-10-19:
hi I can not recieve image as out put of: Camera.listen() 
would you help me?