carla-simulator/carla #3815: IMPORT Radar Model in AirSim

tayabalik, 2021-01-20:
Hello, 
I am trying to build a radar model using Microsoft AirSim which runs on Unreal Engine. Is it possible to use the Radar model from Carla on AirSim. Any leads on the implementation of a Radar model would be appreciated.

germanros1987, 2021-03-11:
I think this question belongs in the AirSim repository.

closed 2021-03-11 by germanros1987

ZifaZhu, 2021-07-25:
Hello, have you implemented this function?

NedaRH, 2022-12-02:
Hi tayabalik, have you implemented this function?

tayabalik, 2022-12-02:
Hello @NedaRH @ZifaZhu , I created a model using AirSim where I used the points from the Lidar function on AirSim and simulated it as Radar using a clustering algorithm DBSCAN.

NedaRH, 2022-12-02:
Hi tayabali,

Thanks for replying and consideration my request. Can I ask if is possible you share your code? Honestly I am also want to use AirSim for radar.

Best,
Neda

[△EXTERNAL]

zhaozhiqing1, 2023-11-29:
Hi @tayabalik ,

Can I ask if is possible you share your code? Honestly I am also want to use AirSim for radar.

Best,
Zhaozhiqing1