carla-simulator/carla #3817: problem in ./GenerateProjectFiles.sh while building Unreal Engine

ammaryasirnaich, 2021-01-20:
I am following the instructions detailed in  the [official link](https://carla.readthedocs.io/en/latest/build_linux/).
When I run this command in the terminal **./GenerateProjectFiles.sh**, the following problem happens:

root@924c5360c26f:~/UnrealEngine_4.24# ./GenerateProjectFiles.sh
Attempting to set up UE4 pretty printers for gdb (existing UE4Printers.py, if any, will be overwritten)...
	updated UE4Printers.py
	found necessary entries in ~/.gdbinit file, not changing it.

Setting up Unreal Engine 4 project files...

Fixing inconsistent case in filenames.
Setting up Mono
ERROR: Unable to instantiate instance of 'BenchmarkToolTarget' object type from compiled assembly 'UE4ProgramRules'.  Unreal Build Tool creates an instance of your module's 'Rules' object in order to find out about your module's requirements.  The CLR exception details may provide more information:  System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> UnrealBuildTool.BuildException: GetBuildPlatform: No BuildPlatform found for Linux
         at UnrealBuildTool.UEBuildPlatform.GetBuildPlatform (UnrealBuildTool.UnrealTargetPlatform InPlatform, System.Boolean bInAllowFailure) [0x0003e] in <93753aa1f6234207acde1540c2b9495c>:0 
         at UnrealBuildTool.TargetRules..ctor (UnrealBuildTool.TargetInfo Target) [0x002fd] in <93753aa1f6234207acde1540c2b9495c>:0 
         at BenchmarkToolTarget..ctor (UnrealBuildTool.TargetInfo Target) [0x00000] in <13ab9fd4eec746a2b3b3e7298e58b27f>:0 
         at (wrapper managed-to-native) System.Reflection.MonoCMethod.InternalInvoke(System.Reflection.MonoCMethod,object,object[],System.Exception&)
         at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00005] in <98fac219bd4e453693d76fda7bd96ab0>:0 
          --- End of inner exception stack trace ---
         at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00017] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.Reflection.MonoCMethod.DoInvoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0007a] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.Reflection.MonoCMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0002f] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.Reflection.MethodBase.Invoke (System.Object obj, System.Object[] parameters) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at UnrealBuildTool.RulesAssembly.CreateTargetRulesInstance (System.String TypeName, UnrealBuildTool.TargetInfo TargetInfo) [0x00093] in <93753aa1f6234207acde1540c2b9495c>:0

I have tried the solution mentioned in the [link](https://github.com/microsoft/AirSim/issues/2829) but i didn't helped. 
I had also tried to install the docker version but the same error appears.
I'm installing it on a remote server (in a docker). Its OS is Ubuntu 18.04. 
I will really appreciate that if someone can help me with this issue.
Thanks

yuemingl, 2021-02-26:
I believe there are some files are broken during the Setup.sh step. Just checkout the repo again or clean up the files and run Setup.sh again will fix the problem.

ammaryasirnaich, 2021-03-02:
Thanks @yuemingl, but I have managed to run Carla docker's image.

closed 2021-03-02 by ammaryasirnaich