carla-simulator/carla #3821: Vehicles Spawn in the air whether on a default map or a map from RoadRunner

5buy5bobee, 2021-01-22:
Hello, I tried to import a vehicle to the project using a default map, but my vehicle floated in the air. However if I tried another vehicle like mini, it worked perfectly. I have followed the steps from CARLA tutorials and checked them lots of times. How could I fix this problem? Thank you!

I read the tutorials to export a vehicle from blender.
https://continuebreak.com/articles/how-rig-vehicle-blender-28-ue4/

And I followed these steps to import to CarlaUE4.
https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/

Here is what we see in Client. It looks like on the ground but it actually floats in the air.
[image]

The current progress is that the throttle, steer, brake seems working, but It can't move forward.

5buy5bobee, 2021-01-22:
I have read this but I do change the physics type to Kinematic.

https://github.com/carla-simulator/carla/issues/3821

XGodina, 2021-01-27:
Hi, @5buy5bobee 
Do you still have the problem?
Do you change the shape radius of wheels?  If the radius is greater than your wheel it will float
[image]

5buy5bobee, 2021-02-09:
> Hi, @5buy5bobee
> Do you still have the problem?
> Do you change the shape radius of wheels? If the radius is greater than your wheel it will float
> [image]

Yes I had tried it before but it had not worked then. Still a big thank to you!

5buy5bobee, 2021-02-09:
I have solved the problem since I found I set general collider as "Kinematic" instead of "default".

After I changed the settings the vehicle works now.

XGodina, 2021-02-10:
Great I glad to hear that !!!

closed 2021-02-10 by XGodina