carla-simulator/carla #3823: git apply error (win10,carla0.9.11)

SunHao-TSU, 2021-01-23:
Hi, i am building carla0.9.11 on windows 10. 
but got errors after **git apply UE4_patch_wheels.patch**" .

_error: patch failed: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/PhysXVehicleManager.cpp:4
error: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/PhysXVehicleManager.cpp: patch does not apply
error: patch failed: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/PhysXVehicleManager.h:99
error: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/PhysXVehicleManager.h: patch does not apply
error: patch failed: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/WheeledVehicleMovementComponent.h:11
error: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/WheeledVehicleMovementComponent.h: patch does not apply_

Wish any help about this issue!

elvira3871, 2021-02-22:
I also encountered the same mistake, have you solved it?

YuqiaoBai, 2021-03-04:
I have the same issue

corkyw10, 2021-03-11:
@DSantosO could you have a look at this please?

AdMaitre1, 2021-04-12:
I'm having a similar issue anyone has a clue on how to solve this ?

My errors are :
error: patch failed: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/PhysXVehicleManager.cpp:4
error: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/PhysXVehicleManager.cpp: patch does not apply
error: patch failed: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/PhysXVehicleManager.h:99
error: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/PhysXVehicleManager.h: patch does not apply
error: patch failed: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/WheeledVehicleMovementComponent.h:11
error: Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Public/WheeledVehicleMovementComponent.h: patch does not apply

DSantosO, 2021-04-13:
Hello,
I have trie to reproduce your problem but for me it works good in Windows 10, could it be some problem with some white spaces? 
Have you try some git options that could fix? Something this like:
`git apply --reject --whitespace=fix git apply --reject --whitespace=fix UE4_patch_wheels.patch`
`git apply --ignore-space-change --ignore-whitespace UE4_patch_wheels.patch`

AdMaitre1, 2021-04-13:
Hello, i just tried the commands you sent me and it displays me the same error as with using git apply  UE4_patch_wheels.patch --verbose . The problem is related to some header files in Engine/Plugins/Runtime/PhysXVehicles/Source/. Can it come from the patch version i use? (used the link in the tutorial at "https://carla.readthedocs.io/en/latest/build_windows/" which is "https://carla-releases.s3.eu-west-3.amazonaws.com/Backup/UE4_patch_wheels.patch" and UnrealEngine 4.24 as sugested)

The complete error message is in the .pdf linked to this message
[Error_git_apply_whitespacefix.pdf](https://github.com/carla-simulator/carla/files/6302717/Error_git_apply_whitespacefix.pdf)

Thanks for your reply, if you have any other possible solutions i'll be glad to try them.

DSantosO, 2021-04-13:
I see,
Can you go to the file ` Engine/Plugins/Runtime/PhysXVehicles/Source/PhysXVehicles/Private/PhysXVehicleManager.cpp` and copy here the first 30 lines?

elvira3871, 2021-04-13:
I have been lucky to solve this problem, but based on the successful installation of unreal. I have two computers. When the first computer was installed successfully, the second computer encountered this error. I moved the entire unreal package in the successfully installed computer to the second computer and continued with the next step.

DSantosO, 2021-04-13:
Hello @elvira3871,
I am happy to hear that.
Do you know why it was not successful on the second computer? Could it be that you had a problem when cloning the repository or that was not in the correct Unreal Engine branch?

AdMaitre1, 2021-04-13:
This is what i have in PhysXVehicleManager.cpp from Private folder: 

// Copyright 1998-2019 Epic Games, Inc. All Rights Reserved.

#include "PhysXVehicleManager.h"
#include "UObject/UObjectIterator.h"
#include "TireConfig.h"

#include "Misc/CoreMiscDefines.h"
#include "PhysicalMaterials/PhysicalMaterial.h"
#include "Physics/PhysicsFiltering.h"
#include "PhysXPublic.h"
#include "Physics/PhysicsInterfaceCore.h"

DEFINE_LOG_CATEGORY(LogVehicles);

#if WITH_PHYSX_VEHICLES

//Angle thresholds used to categorize contacts as suspension contacts or rigid body contacts.
#define POINT_REJECT_ANGLE PxPi/4.0f
#define NORMAL_REJECT_ANGLE PxPi/4.0f

DECLARE_STATS_GROUP(TEXT("PhysXVehicleManager"), STATGROUP_PhysXVehicleManager, STATGROUP_Advanced);
DECLARE_CYCLE_STAT(TEXT("PxVehicleSuspensionRaycasts"), STAT_PhysXVehicleManager_PxVehicleSuspensionRaycasts, STATGROUP_PhysXVehicleManager);
DECLARE_CYCLE_STAT(TEXT("PxUpdateVehicles"), STAT_PhysXVehicleManager_PxUpdateVehicles, STATGROUP_PhysXVehicleManager);
DECLARE_CYCLE_STAT(TEXT("UpdateTireFrictionTable"), STAT_PhysXVehicleManager_UpdateTireFrictionTable, STATGROUP_PhysXVehicleManager);
DECLARE_CYCLE_STAT(TEXT("TickVehicles"), STAT_PhysXVehicleManager_TickVehicles, STATGROUP_PhysXVehicleManager);
DECLARE_CYCLE_STAT(TEXT("VehicleManager Update"), STAT_PhysXVehicleManager_Update, STATGROUP_PhysXVehicleManager);
DECLARE_CYCLE_STAT(TEXT("Pretick Vehicles"), STAT_PhysXVehicleManager_PretickVehicles, STATGROUP_Physics);

bool FPhysXVehicleManager::bUpdateTireFrictionTable = false;
PxVehicleDrivableSurfaceToTireFrictionPairs* FPhysXVehicleManager::SurfaceTirePairs = NULL;
TMap<FPhysScene*, FPhysXVehicleManager*> FPhysXVehicleManager::SceneToVehicleManagerMap;
uint32 FPhysXVehicleManager::VehicleSetupTag = 0;

DSantosO, 2021-04-13:
Hello @AdMaitre1,
At least in those lines, the patch seems to be correctly applied. Can be that you are trying to apply it twice?
If the patch is correctly applied, there should be three files staged for commit when doing a `git status`. 
If you checkout the three of them, you can apply it again to be sure but always with a clean `git status`.

AdMaitre1, 2021-04-13:
Yes i have typed the command serveral times as i had to reboot my computer. I was thinking it didn't applied correctly as i had those errors and also further errors on the build lead me to think it came from this patch. The Setup.bat and GenerationPojectFiles.bat were successfull so i think the patch applied. 
I'm carrying on the build, letting you know if i finished without wrong outputs. Thanks for your replies!

closed 2021-05-26 by DSantosO

keeees, 2021-06-03:
> Hello @AdMaitre1,
> At least in those lines, the patch seems to be correctly applied. Can be that you are trying to apply it twice?
> If the patch is correctly applied, there should be three files [...]

I had the same problem and I was able to resolve it by   git reset --hard.