carla-simulator/carla #3825: Autopilot not working with multiple actors

j-o-d-o, 2021-01-24:
Carla Version: 0.9.11
OS: Linux Mint 19

I am trying to generate some random dense traffic. Urban and on Highway. But I encountered some issues when I spawn many actors.
1) Most vehicles do not get spawned at some point (after around 60) because it seems the spawn points from `get_spawn_points()` are all occupied and it would result in collisions.
1) The spawn points seem to not always take into account to spawn the vehicle on the correct side of the road
2) Many of the spawned actors just don't move at all despite having autopilot set to `True` and then everybody is just standing (including the ego vehicle)

Is there some limitation to this? Or a more novel approach to have randomly generated (dense) traffic?

code snipet:
```python
  self.vehicle_npcs = []

  for i in range(300):
      bp = random.choice(blueprint_library.filter('vehicle'))
      spawn_point = random.choice(world.get_map().get_spawn_points())
      npc = world.try_spawn_actor(bp, spawn_point)
      if npc is None:
          print(str(i) + " :Failed spawning...")
      else:
          self.vehicle_npcs.append(npc)
          npc.set_autopilot(True)
```

Now, it's a bit hard to show "not moving" with a screenshot. You just have to take my word that in this scene everything ist standing still :D
[image]

jackbart94, 2021-01-25:
Hi,

Check our ``spawn_npc.py`` script to see how to spawn NPC vehicles correctly!

j-o-d-o, 2021-01-26:
Sweet, thx. Totally overlooked that when browsing the examples.

closed 2021-01-26 by j-o-d-o

JieruiLin, 2021-07-25:
@j-o-d-o I'm facing the same issue. Wonder how you fixed that? Thanks!

hailuu684, 2021-08-12:
Do you know how to get information from those vehicles? They are using command to spawn, but I do not see any method to retrieve the information like velocity, acceleration,...

marychrisgo, 2022-04-03:
@aeon0 I'm currently facing the same issue. I am saving depth and RGB images but the npcs are not moving though I set the autopilot to be True. How did you fix it? Thanks!