carla-simulator/carla #3827: A caral server cannot use multiple GPUs at the same time

yuexishuihan, 2021-01-26:
#### A caral server cannot use multiple GPUs at the same time.
carla 0.9.11 docker+nvidia docker mod.
I can use one of the GPUs, which I can control, but I can’t use more than one at the same time, and the current GPU utilization rate is not high, only 30-40%.
command
```
sudo docker run -d -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=0 carlasim/carla:0.9.11 /bin/bash -c 'SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -ResX=800 -ResY=600 -nosound -windowed -opengl'
```
NVIDIA_VISIBLE_DEVICES
Does anyone have the same problem as me？
Platform:Ubuntu 20.04.1 LTS
Carla version: 0.9.11
NVRM version: NVIDIA UNIX x86_64 Kernel Module  460.32.03  Sun Dec 27 19:00:34 UTC 2020
1d:00.0 VGA compatible controller: NVIDIA Corporation Device 2204 (rev a1)
1d:00.1 Audio device: NVIDIA Corporation Device 1aef (rev a1)
b4:00.0 VGA compatible controller: NVIDIA Corporation Device 2204 (rev a1)
b4:00.1 Audio device: NVIDIA Corporation Device 1aef (rev a1)
-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 3090    Off  | 00000000:1D:00.0 Off |                  N/A |
| 34%   30C    P8    30W / 350W |   1027MiB / 24268MiB |     37%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   1  GeForce RTX 3090    Off  | 00000000:B4:00.0 Off |                  N/A |
| 32%   27C    P8    25W / 350W |      5MiB / 24268MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2000      G   /usr/lib/xorg/Xorg                  4MiB |
|    0   N/A  N/A     26262    C+G   ...x/CarlaUE4-Linux-Shipping     1017MiB |
|    1   N/A  N/A      2000      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+

qhaas, 2021-01-30:
I have noticed that a single carla process cannot effectively use multiple GPUs at once.  I'm not sure if [SLI](https://docs.unrealengine.com/en-US/RenderingAndGraphics/Nvidia/NVIDIAAlternateFrameRendering/index.html) is supported on Linux like it is on Windows with the underlying UnrealEngine, which would explain this.

Worse still, it will not automatically 'load balance' on multi-gpu systems and will just throw multiple server runs onto gpu 0, even if gpu 0 is pegged at 100% and the other GPUs are idle.  As you discovered, you can use [docker](https://carla.readthedocs.io/en/latest/adv_rendering_options/#docker-recommended-approach) or [SDL environment variables](https://github.com/carla-simulator/carla/issues/225) (outside of docker) to manually restrict it when running with OpenGL.  Unfortunately, using environment variables appears broken with vulkan, at least when using UE 4.25 (which, to be fair, is unsupported by CARLA).

To be fair, I think this is more of an UnrealEngine thing than a CARLA thing.

yuexishuihan, 2021-02-02:
Okay, thank you for your answer. The bad thing is that it seems impossible to use multiple GPUs in a short time.

qhaas, 2021-02-02:
Might want to ask over at [UnrealEngine Answers](https://answers.unrealengine.com/index.html), they recently helped me with UnrealEngine GPU selection questions.

yuexishuihan, 2021-02-03:
Okay.This is a good idea, I will try

corkyw10, 2021-03-11:
@bernatx could you have a look at this please?

qhaas, 2021-04-13:
@yuexishuihan , did you discover anything about leveraging multiple GPUs for CARLA?  The closest we have come is just running multiple client/servers on the same GPU (i.e. larger number of independent simulations).

yuexishuihan, 2021-04-27:
Not yet, I am currently exploring adding/modifying assets in carla, including cars,
Maps, weather, etc.

jimmyw404, 2021-08-14:
I'm very interested in multi GPU or multi computer CARLA simulation! If anyone has gotten multippe gpus to work or has tried to distribute the carla computing please let us know!!!!

gouchaoer, 2021-09-29:
any progress?
I am doing realtime 4 camera output in HIL, and I found it's impossible

jimmyw404, 2022-01-09:
I'm still very interested in this! Unreal doesn't seem to handle multi-GPU simulation very well, so solving this problem is non-trivial.

jimmyw404, 2022-04-16:
Still very interested in this feature and looking forward to https://github.com/carla-simulator/carla/tree/bernat/multi_gpu !!!

closed 2024-07-05 by Blyron