carla-simulator/carla #3828: Behaviour agent stops permanently at 'stop sign' in an uncontrolled junction.

karthi0804, 2021-01-26:
Hi, I am facing a problem when using a behavior agent in Carla 0.9.11 in Ubuntu 18.04. The agent stops permanently at an uncontrolled junction located roughly at (143, 2), town 05.  The expected behavior is to wait for a while and cross the junction when the junction is free of vehicles. Please help me if I am missing something to configure.

Esquilli, 2021-02-21:
@karthi0804 Did you figured out this issue? My "fix" was to set `ignore_tracffic_light` to `True`

karthi0804, 2021-02-22:
> @karthi0804 Did you figured out this issue? My "fix" was to set `ignore_tracffic_light` to `True`

I didn't find the root cause. I also used the same workaround.

corkyw10, 2021-03-11:
@glopezdiest could you have a look at this please?

GarethZhang, 2021-05-27:
Hi @karthi0804 @Esquilli ,
I am facing the same problem here. Where can I find the option to set `ignore_traffic_light` to True?
Thanks!

Esquilli, 2021-05-27:
@GarethZhang When you initialize an agent (i.e BehavioralAgent) you can set the `ignore_traffic_light` parameter to True like this `BehavioralAgent(ignore_traffic_light=True)`.

Check the class init parameters [here](https://github.com/carla-simulator/carla/blob/4c8f4d5f191246802644a62453327f32972bd536/PythonAPI/carla/agents/navigation/behavior_agent.py#L36).

GarethZhang, 2021-05-29:
@Esquilli Thanks a lot!

bsande6, 2021-11-18:
Has anyone found a solution that allows the agent to still follow normal traffic rules?

glopezdiest, 2021-12-01:
@bsande6 Are you using 0.9.12 or higher? As far as I remember, the issue is that the behavior agent uses the `vehicle.is_at_traffic_light` function, which always returns True at stop signs, hence why the vehicle stops permanently. I think we fixed it for 0.9.12 though, but I may be wrong. I'll test it ASAP, and report back

bsande6, 2021-12-01:
@glopezdiest  I was using 0.9.11 at the time but have upgraded since and solved the problem.

closed 2021-12-02 by glopezdiest