carla-simulator/carla #3829: Building is not visible by LIDAR

zygfrydw, 2021-01-26:
Carla Version: 0.9.10
Platform:Ubuntu 20.04.1 LTS
Tested in ROS2 Foxy with carla-bridge. 

Lidar is passing through some buildings in `Town03`, not detecting a building wall, although the car is colliding with the building.
The white points are point cloud map recorded in Carla 0.9.9, the color points are points registered by 2 lidars.

Building 1
[image]

[image]
[image]
[image]

Building 2
[image]
[image]

XGodina, 2021-02-19:
Hi, 

Sorry for the late answer. I will take look more deeply at it. Thanks to reported

XGodina, 2021-03-17:
Hi @zygfrydw We know what happens there with the lidar and now we are working to fix it

zygfrydw, 2021-03-17:
Great, thanks for your help!

XGodina, 2021-03-19:
Hi @zygfrydw, We've discovered the problems. Some Blueprint in our buildings, there are with a wrong collision preset. Now we add a correct collision preset and now appear in the lidar. The PR is waiting to accept.

yuvalHG, 2021-09-12:
Is there any new with this issue? is there any branch merged with those changes?

closed 2023-06-02 by stale[bot]