carla-simulator/carla #3834: Carla 0.9.11 crashes without any error reporting in docker

zxqzhang, 2021-01-27:
Hi, I'm using carla docker (version 0.9.11) on my ubuntu server (ubuntu 18.04, with Nvidia driver version 460.36, and I have also tried with other driver versions). The command $ ./CarlaUE4.sh in the docker bash just crashes and does not report any information except for the following regular lines:

4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found

Then it just stops and nothing happens (moving on to the next command line). Could you please help me with it?

qhaas, 2021-01-27:
A few thoughts since this could be a number of things...
1. Did you deploy [nvidia-docker](https://carla.readthedocs.io/en/latest/build_docker/#nvidia-docker2), this is separate from the driver install
2. Does `nvidia-smi` inside the container return anything meaningful (if not, nvidia-docker likely isn't setup correctly)?
3. Have you tried adding the `-opengl` flag to see if it is a [vulkan issue](https://carla.readthedocs.io/en/latest/adv_rendering_options/#vulkan-vs-opengl)
4. Did you start docker [as they recommend](https://github.com/carla-simulator/carla/blob/0.9.11/Util/Docker/Release.Dockerfile#L7):
```
# If you want the highest performance you need to make sure that your host has a functioning Xorg server running
# even in headless/cloud systems.
# If your host has a running version of Xorg then set  SDL_VIDEODRIVER=x11 and run the docker container as:
#
# sudo -E docker run --privileged --rm --gpus all -it --net=host -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw carla:latest /bin/bash
#
# Otherwise, set SDL_VIDEODRIVER=offscreen and run the docker container as:
#
# sudo -E docker run --rm --gpus all -it --net=host carla:latest /bin/bash
```

zxqzhang, 2021-01-28:
> A few thoughts since this could be a number of things...
>
> 1. Did you deploy [nvidia-docker](https://carla.readthedocs.io/en/latest/build_docker/#nvidia-docker2), this is separate from the driver [...]

1. Of course I did.
2. nvidia-smi inside the container works fine
3. I have tried the -opengl flag in my containers, nothing different happened
4. I started the docker a little differently, I started a bash console instead of directly starting the CarlaUE4.sh, but I believe it makes no difference.

zxqzhang, 2021-01-28:
We rechecked the SDL_VIDEODRIVER setting, and it works, thank you!

closed 2021-01-28 by zxqzhang

maverick108, 2022-11-28:
Where to see the carla error logs?