carla-simulator/carla #3835: Errors when building RSS dependencies

zhuhj-tery, 2021-01-28:
Hello CARLA team,

I want to use the RSS sensor in CARLA platform and encountered some problems:
1. When using the pre-packaged version of CARLA, the RSS sensor could be attached to the vehicle. However, the callback function does not take effect and there is no output from the sensor. Is it because the pre-packaged version of CARLA does not have the intel-ad-rss libraries? Can I do something with the pre-packaged version of CARLA to integrate the intel-ad-rss libraries?
2. I also try to build CARLA from the source code. I first followed the guideline of the Linux building of CARLA and some examples have been used to test the functionality. Then, I want to build the full set of dependencies and RSS components. The command "make LibCarla.client.rss" works well, while the command "make PythonAPI.rss" fails. The error message is shown below:
[image]
[image]
[image]
It seems that some files are lost. Is there any possible solution to this problem?

Thank you very much for your time.

corkyw10, 2021-03-11:
@fpasch could you follow up on this please?

berndgassmann, 2021-03-11:
1) Are you sure, you downloaded the RSS Variant of CALRA?
Because these come with pre-compiled RSS and contain the RSS-library (your linker error of the build is showing that exactly these libraries being linked are not there on your build, but usually these are ;-)
https://github.com/carla-simulator/carla/releases
And those should work.

2) I don't know what failed exactly, but when I look into the output I see that
make PythonAPI.rss
was not your first build call.
Maybe a previous build without RSS didn't copy the required files. Or a build of the RSS dependency actually failed. The check done in Ad-rss.sh is simplistic: if the Build/ad-rss-4.0.0/install folder is there and build folder for the correct python version, too, it assumes all was done successful to speed up the consecutive builds for the vast majority. And the carla Python library build requires the LibCarla client build to copy files just over. So also there no meaningful checks are done if all is found and was copied correctly...
So maybe this fixes the build or at least shows the actual problem you face:
rm -rf Build/ad-rss-4.0.0/install
rm -rf PythonAPI/carla/dependencies
make PythonAPI.rss

The first should ensure that "colcon" build system is again triggered for the RSS-parts and should install all things that already should have been built.
The second ensures that the copy-over step as basis for the python build is done within "LibCarla ClientRSS.Release build".

On a system already compiled correctly the above is mainly copying stuff around and linking the python library.

Hope this helps.

zhuhj-tery, 2021-03-11:
Thanks! It helps.

closed 2021-03-11 by zhuhj-tery