carla-simulator/carla #3839: Traffic manager not deterministic across runs in Carla 0.9.11

pavanteja295, 2021-01-28:
Hello, 

I am trying to generate RGB images using Carla 0.9.11. I want to maintain deterministic behaviour  across runs. I see that the new release fixed the non-determinism of Traffic Manager. However, I still find the Traffic Manager to be random across runs after some frames(around 3000 frames).  I followed spawn_npc.py script and the only change I made is the following:
 
`fixed_delta_seconds=0.5 and settings.max_substeps = 15 and settings.max_substep_delta_time = 0.08` 

The simulator is consistent ~ 300 ticks but not after that. Can you please let me know if this is a bug?  

of the simulator across different runs. What I observed is that after some frames rgb outputs of any two runs

jackbart94, 2021-02-17:
Sorry about the delay answering this issue, it's actually very important to us. First off, a few questions:
Have you been able to fix it since you opened the issue?
Were you using ``synchronous_mode`` (by using ``--sync`` in ``spawn_npc.py``) and setting correctly the seed?
Are you on Linux or Windows?

pavanteja295, 2021-02-23:
Hello, yes I used the synchronous_mode infact the simulation is synchrous for some simulation after which it starts behaving non-deterministic. I also made sure to set the traffic manager seed and other sync options. I can share with you more details if you would like to know more. It's on Ubuntu 18.04 64 bit system

jackbart94, 2021-02-23:
So, I have been running the ``spawn_npc.py`` with determinism, synchronous mode and your custom values.

From my side, I can tell you that it's best to keep the ``fixed_delta_seconds`` to ``0.05`` because it's what Traffic Manager and the physics engine are using, which could be why it is breaking determinism.

Regarding the other parameters, @DSantosO can probably tell you more whether it's wise to change them or not, in terms of determinism, because even with ``fixed_delta_seconds`` to ``0.05`` it still seems to be not deterministic with the other changes.

On a side note, I have ran the ``spawn_npc.py`` script with those values and cars aren't moving in the proper way. Not sure if you've personally taken a look at it, but that's probably not what you're looking for.

DSantosO, 2021-02-23:
I agree with @jackbart94, the fixed_delta_seconds is too high, it should be `0.05` or at most `0.1`. We also recommend `max_substep_delta_time`to be below `0.0166666`.

pavanteja295, 2021-02-25:
@jackbart94  @DSantosO  I think its not just with the above recommended parameter values the simulator turns non-deterministic but just that we  have to **wait longer** because the delta seconds is set to much lower now (0.05). May be try with the recommended values and wait for a longer time and let me know you still see this problem. For me its still non-deterministic.

jackbart94, 2021-03-10:
> @jackbart94 @DSantosO Even with the recommended parameter values the simulator turns non-deterministic but just that we have to **wait longer** because the delta seconds is set to much lower now (0.05). May be try with the recommended values and wait for a longer time and let me know you still see this problem. For me its still non-deterministic.

We have several tests running this and they are all deterministic. I also have tried with an RBG camera and 10000 ticks, and it came out correct. Are you still changing the settings.max_substeps and settings.max_substep_delta_time? Can you try with the default ones, or with those that @DSantosO suggested?

pavanteja295, 2021-03-16:
I shall create a simple script taken from what I use and shall upload it here. Doing a lot runs indeed resulted in some nondeterministic results because I was not changing the parameters you mentioned and used the default params. May be you can try to reproduce and see if you are getting deterministic values.

jackbart94, 2021-03-17:
> I shall create a simple script taken from what I use and shall upload it here. Doing a lot runs indeed resulted in some nondeterministic results because I was not changing the parameters you [...]

I've tried with any of your parameters and I didn't get deterministic results, but this is because all of your changes are outside the bounds of acceptable values to have determinism. If I instead use the standard default CARLA values and just add an RGB camera, I have seen determinism always, even after 10.000 steps.

pavanteja295, 2021-04-03:
@jackbart94  I'm not sure how you are checking the determinism but I'm tried to quantatively  compare images. I used to the recommended settings and I found an instance like this: 

I tried to sample 10 scenes with 5000 total simulation runs. This experiment has been repeating multiple times. Out of 50 images I collected 10 images had some problems. These 10 are highly correlated and happened at once instance as once the traffic manager is non-deterministic the future events also cannot be same across runs.

[image]
[image]

jackbart94, 2021-04-03:
Can you share the script that you're using, without the image generation? 

We are checking determinism using the CARLA recorder, you can see an example in our smoke tests on how to do that, it should be called test_determinism.py

pavanteja295, 2021-04-03:
I can create the condensed version of the data gen script and get back to you? 
I'm confident that I'm setting the params correctly because the traffic manager is "almost" deterministic just in a  few instances it would not be. Sure I shall check the script you mentioned.

closed 2021-11-18 by jackbart94

Christin95, 2023-03-09:
@jackbart94 I#m experiencing the same issue. I am on Ubuntu 18 and using Carla 0.9.14. I can see deterministic results in the first minute or so of the simulation but then it starts to behave random. I have the following world settings:
`synchronous_mode=True,no_rendering_mode=False,fixed_delta_seconds=0.05,substepping=True,max_substep_delta_time=0.01,max_substeps=10,max_culling_distance=0,deterministic_ragdolls=True`
I am confident the traffic manager and seed key are initialized correctly.