carla-simulator/carla #3841: 'TrafficManager' object has no attribute 'global_distance_to_leading_vehicle'

ll7, 2021-01-30:
## Error

```
Exception has occurred: AttributeError
'TrafficManager' object has no attribute 'global_distance_to_leading_vehicle'
  File "/home/luttkule/git/carla_optimization/simple_scenario.py", line 15, in <module>
    tm.global_distance_to_leading_vehicle(1)
```
## code to replicate
```python
import carla
import time

client = carla.Client('localhost', 2000)
client.set_timeout(10.0) # seconds
# client.reload_world()

world = client.get_world()
world = client.load_world('Town03')

# traffic manager
tm = client.get_trafficmanager(8000)
tm.set_random_device_seed(0)
tm.global_percentage_speed_difference(-5.0)
tm.global_distance_to_leading_vehicle(1)
...
```

## Software Version
os: ubuntu 20.04
python: 3.8.5
carla: 0.9.10.1 binary download

ll7, 2021-01-30:
https://github.com/carla-simulator/carla/blob/0c97e9a5de5b35b759cc5b0955801244ed76791f/PythonAPI/examples/spawn_npc.py#L112

uses a set method, which is not listed in the python api reference and not in the example.

jackbart94, 2021-03-10:
It's fixed now, but you are checking the 0.9.10v of documentation. We will make sure that one is correct as well.

closed 2021-03-25 by jackbart94