carla-simulator/carla #3843: Add Vehicle - UE measurements don't change regardless of scale in Blender

ancabilloni, 2021-01-31:
Hello,
I have been issue with scaling with importing fbx model from Blender into Unreal Engine. The actual measurement doesn't seem to change despite whatever size (or scale?) of the vehicle. Please see two examples below:

In this example, I use 1.0 scale in Blender
[image]
[image]

Keeping everything the same as 1st example, I changed the scale to 0.01 in Blender
[image]
[image]

The numbers show in UE4 is the meter value in Blender. Checking other cars in Carla, I believe it's supposed to show cm in UE4? Due to this issue, I believe I have been getting spawn collision error. 

Any pointer for what I did wrong is greatly appreciated. Thanks!

ancabilloni, 2021-01-31:
I resolved this issue. I decided to started from scratch. I set the scene unit to 0.01 scale before import anything. After that I import the Skeleton and the car model (supposedly in their true scale). 

I think i messed up the scaling way too many times by keep changing the scene scaling with the models already in it. And changing scale of the skeleton and model itself. 

So my recommend for those who face the same issue is:
1. Prepare your model separately with scale 1 at its true measurements in meter, and correct orientation for Unreal Engine (x-forward, z-up)  and save to fbx file. 
2. Start a new blender environment > Set Scene unit to 0.01 scale
3. Import the General4WheeledVehicleSkeleton fbx file
4. Import the Vehicle fbx file prepare in step 1.
5. Work on the usual rigging and export.

ancabilloni, 2021-01-31:
There still seems like some physics issue with the performance still. 

Here is a sample clip. You can see the car is a bit "hovering" ?  and when the car brakes, the front sort of tips up.

https://user-images.githubusercontent.com/23693651/106399655-fe4d3500-63e7-11eb-8698-bf267da6981b.mp4

Here is what I have for the vehicle in PhysicsAsset
[image]
[image]

parth3153, 2021-02-14:
did you upscaled in blender and then started rigging, if you didnt thats causes this problem, i have read your text above with the scaling isuue and i was struggling aswell, the i used the ue4 addon for Blender to rig vehicles its works magic and fast process, the vehicle turns out perfect in carla, check continuebreak.com for the addon

ancabilloni, 2021-02-14:
@parth3153 I went through the same tutorial for get the rigging to work, I used the addon as well. But eventually just do manual rigging, well, just for the fun of it.
The scaling issue was a bit tricky. But because I need to work on several import iterations, I stick with the routine mentioned above to get it to work in every iteration now.

closed 2021-02-14 by ancabilloni