carla-simulator/carla #3844: Simulate holes in the ground

alinakir, 2021-02-01:
Hi, I want to simulate a hole in the ground (lets say 0.5mx0.5mx0.5m, can be round can be rectangle, doesn't matter) to see how lidar points hit it. I didn't find anything about how to make holes/pits/sinkholes in the ground/road. 

any ideas?

jimmyw404, 2021-02-05:
I'm also interested in this. My guess is that this isn't the responsibility of carla or even unreal and should instead be done at the 3d modeling level, ex blender. I don't know of any good tools built on this, but i know there are a variety of tools looking at road surface simulation.

https://en.wikipedia.org/wiki/OpenCRG is one.

XGodina, 2021-02-19:
Hi @jimmyw404 @alinakir 

We have some assets that you can spawn on road. It may help. But we don't have holes yet.

closed 2023-06-02 by stale[bot]