carla-simulator/carla #3845: Possibility to have IMU & GNSS sensor rate higher than frame rate (camera sensor)

goksanisil23, 2021-02-01:
Hi,
My goal is to get 20Hz lidar, 100Hz gnss and 100Hz imu data over ros bridge from Carla. In versions 0.9.10, 0.9.11, I've tried setting in sensors.json file the corresponding sensor_tick values.  However, these don't seem to affect the publish rate of these sensors, since I keep seeing 20Hz on rostopic hz. I believe this is due to the fact that I'm running with fixed_delta_seconds 0.05.

To get 100Hz from IMU and GNSS, I can change the fixed_delta_seconds to 0.01, which makes all sensors calculate at 100Hz.  (still, sensor_tick has no affect on publish rate)

Is the rate of Unreal simulation step need to be determined by the fastest sensor in our setup? If so, setting 0.01 fixed step time and then lowering the sensor_tick for lidar to 0.05 should work for lidar (yet it does not at the moment)

what am I missing here?
Thanks

corkyw10, 2021-03-12:
@DSantosO could you have a look at this please?

DSantosO, 2021-03-12:
Hello @goksanisil23,
The sensors cannot tick at a higher rate than the simulator so you are right in your first conclusion. The second way should work but there was a bug in version 0.9.11 about the sensor_tick. This was fixed in PR #3818 so you can get that to work if you compile from source or in our [overnight build](http://carla-releases.s3.amazonaws.com/Linux/Dev/CARLA_Latest.tar.gz).

goksanisil23, 2021-03-13:
Thanks for pointing me to the PR @DSantosO !

closed 2021-03-15 by DSantosO