carla-simulator/carla #3847: Pedestrians always perform crossing in 0.9.11

ndeshp2s, 2021-02-01:
Hello All, I recently updated CARLA from 0.9.9 to 0.9.11 and one thing I noticed is 'set_pedestrians_cross_factor' does not seem to be working any more now? If I try setting pedestrian crossing to zero, even then I see some pedestrians crossing road. I am wondering why is this happening

corkyw10, 2021-03-12:
Posting the answer from @bernatx from the discord channel here for anyone else wondering:

Town01 has some invisible crosswalks where the pedestrian can cross the road, so maybe that is what you see. It is not that the cross factor function is failing, just that with cross factor of 0, in Town01 and Town02 the pedestrian can cross roads at specific places (without any crosswalk texture in place).

@ndeshp2s let us know if you still have any questions regarding this topic

closed 2021-03-12 by corkyw10

raymanfx, 2021-04-22:
Hi @corkyw10, we just ran into the same issue. On top of that, the `set_pedestrians_cross_factor` function is not included in the latest Python API documentation. Would you mind elaborating on those "specific places" some more? Why was it designed this way, how were those "specific places" selected, etc.?

In our setup, we would prefer the pedestrians not crossing the road under any circumstance. Could you point me to the code where those "invisible crosswalks" are implemented? If you prefer to keep this behavior, we'll probably just carry a downstream patch to remove them.