carla-simulator/carla #3848: Does Carla simulator support indoor environment?

JackRen88, 2021-02-02:
Hello，everyone!
Currently developing indoor and outdoor navigation systems, does carla support indoor scene navigation?

qhaas, 2021-02-02:
Some of the towns have tunnels.

JackRen88, 2021-02-03:
Thank you @qhaas   Carla provide an indoor environment model so that the robot can enter the indoor environment and navigate in an unstructured environment?

XGodina, 2021-02-19:
Hi all,

What do you mean by an unstructured environment? off-road? indoor?

KouWeiBin, 2022-06-20:
> Thank you @qhaas Carla provide an indoor environment model so that the robot can enter the indoor environment and navigate in an unstructured environment?

Hi JackRen88, could you please tell me where the indoor environment you mentioned? Thanks

closed 2023-06-02 by stale[bot]