carla-simulator/carla #3853: Vehicle behaviour on Gradient Road

MS012086, 2021-02-02:
Hello All,
We encountered trembling behavior in vehicles on Gradient road, Whether it is Carla vehicle dynamic issue or Gradient roads are not supported in Carla? Could you please help us regarding this issue.

Tested platform:
Carla: 0.9.9.4
Linux: Ubuntu 18.04 LTS
Python 3.7

https://user-images.githubusercontent.com/65219753/106630096-875b9c00-65a1-11eb-8db3-54808ed3a37c.mp4

qhaas, 2021-02-02:
We have noticed similar 'bouncing' when using OpenGL in CARLA 0.9.8 and 0.9.11, but switching to `-vulkan` from `-opengl` seemed to make it go away.  What [rendering options](https://carla.readthedocs.io/en/latest/adv_rendering_options/) were you using?

MS012086, 2021-02-03:
we have noticed similar bouncing issue when switching over to -vulkan from -opengl in carla 0.9.9.4

https://user-images.githubusercontent.com/65219753/106723980-b4ee2700-662d-11eb-82f4-1850b409f143.mp4

jackbart94, 2021-02-04:
Hi,

It says that the video can't be played because the file is corrupt. Could you upload them again and also give more information about what you are executing?

MS012086, 2021-02-04:
Hi, 
We are using a road profile with certain elevation angle and trying to run the default carla vehicle in autopilot mode. We could observe normal motion in the straight road but once the vehicle enters  the road with elevated profile we could notice bouncy movement during its motion. we observed similar behaviour even in curved road with elevation angle. what could be the reason behind this weird behaviour of the vehicle in the elevated road? herewith we are attaching the video again

https://user-images.githubusercontent.com/65219753/106914838-69ba3e00-672b-11eb-9403-95e6ea90c253.mp4

Axel1092, 2021-02-05:
Hi @MS012086, how are you generating the map? Are you using Roadrunner? How long is the road? 
With Gradient roads I assume that you mean the shape of the road cross section, I suppose that roadrunner of the software you are using could create small bumps in the road if this is not perfectly aligned, causing the car to have this weird motion. Can you verify that the road is completely flat and without sudden height differences?

MS012086, 2021-02-09:
Hi @Axel1092 , Yes, We are using roadrunner tool for map generation. The road is about 5 km long. We have checked the road profile in roadrunner and could  see that the height of the road increase gradually without any sudden difference. The road is flat and only its height increases gradually after a certain distance . Herewith attaching a screenshot from roadrunner.
[image]

Axel1092, 2021-03-03:
@MS012086 sorry for the late response. I can only think that this issue is due to small variations in the road geometry at the time of exporting. Carla has no support for superelevation or crossfall (non flat cross section in the road). Perhaps roadrunner is exporting additional objects as well. Also, in the parts where the vehicle has the strange behavior we can see this weird noise in the image which leads me to think that there is an issue with the geometry of the road.
[image]

MS012086, 2021-03-08:
Hi @Axel1092 ,

Thank you very much for your information.

We have removed additional objects( Guardrail and surrounding landscape) from road profile.
And now we could not visualize the bouncing behavior on the road.
Even after removing the additional objects, we could see the noise in the image but it is not affecting the vehicle behavior.

dschambach, 2021-03-12:
We have been trying to debug a very similar situation - a flat three lane road which is 5km - with no elevation change. If terrain is present - vehicles appear to collide with invisible humps positioned on the road. Removing the terrain object (which surrounds the road) fixes this behavior.

dschambach, 2021-03-12:
@Axel1092 - Here is an video that shows the difference/problem. Note the vehicles appear to be colliding with some kind of bumps along the road when terrain object is active. The only difference between the first and second part of the video is the deletion of the terrain mesh from the level. 

https://user-images.githubusercontent.com/16140583/111001829-5b1e0280-8339-11eb-8f2b-54911570d564.mp4

dschambach, 2021-03-15:
A workaround - is to import a generic/flat DEM into RoadRunner and apply it to your scene. This changes the triangulation of the terrain mesh before it is exported, which in our case fixed the problem.

Axel1092, 2021-03-16:
It would seem that roadrunner is creating some geometry under the road that is causing the cars to move like that. Perhaps the terrain is not flat under the road and it overlaps with the road itself?

closed 2023-06-02 by stale[bot]

HariKrishnaPadala, 2024-03-19:
I also observed the same behavior in Carla. The car suddenly jumps at some location on the road mesh. The mesh appears to be smooth (verified in 3D modeling viewers) and still the cars jumps, front wheels first followed by rear wheels as if it hit something. This behavior is only observed with cars moving on the road. Other objects like rolling cylinder/sphere in the same place of the road does not hit anything. When I changed the road mesh to the procedural mesh generated from Carla, the car does not jump and follows smoothly. Can anyone suggest what is going wrong here?

dschambach, 2024-03-19:
Are you using road runner to create scenes? Do those scenes have bridges in
them?

HariKrishnaPadala, 2024-03-19:
Not Roadrunner. I am trying to use the procedural mesh from CARLA. The scene does not have bridge, just a normal road intersection. The car jumps mainly when it is trying to cross a intersection, even though the intersection appears to be smooth.