carla-simulator/carla #3859: Autopilot code location

ym2509, 2021-02-05:
Hello!
In the compiled v0.9.9, where are the source code files for Autopilot? For example, when I run the manual_control.py example in the PythonAPI folder, which files/scripts is it accessing to get the code for autopilot? I removed the basic_agent.py, behaviour_agent.py and roaming_agent.py files from the PythonAPI/carla/agents/navigation folder but autopilot still worked, so I don't think these are being used by the example clients. This is the version I am using: https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.9.tar.gz

All help greatly appreciated!

jackbart94, 2021-02-05:
In CARLA 0.9.9, the autopilot is called Traffic Manager. [Here you can read everything about it.](https://carla.readthedocs.io/en/latest/adv_traffic_manager/#traffic-manager)

Why do you want to remove it? What do you want to achieve?

ym2509, 2021-02-05:
Thanks for replying!
Yes, I have been referring to the page. My main goal is to set the start and end destinations for the ego vehicle so that it follows the same route each time I set it to Autopilot. 
For example, if I took the automatic_control.py client in the example folder and added some code to set the start and end destinations, I would like the car to spawn in the same place and drive to the same location each time the client is run.
Please note I am using the compiled version so I do not have a 'Source' folder or WheeledVehicleAIController.cpp.
Thanks :)

jackbart94, 2021-02-08:
``WheeledVehicleAIController.cpp`` is deprecated. I think you should read more carefully about Traffic Manager from the link I posted before, to understand the situation.

Then, ``automatic_control.py`` uses the scripts in the ``agent`` folder, which aren't related to Traffic Manager, which also is coded in C++ so you can't change it from the compiled version.

Also, there already exists an agent which allows users to set start and end destination, and it's the Behavior agent.

Traffic Manager is meant for vehicles to roam indefinitely around the towns, so if you want to give it a start and end destination, you must read carefully the link I sent you and change parts of the architecture in C++.

ym2509, 2021-02-08:
I read the [v0.9.9 page](https://carla.readthedocs.io/en/0.9.9/python_api/) again, and I think I have a better understanding now after your explanation.

Just to confirm my understanding, if you don't mind, if I have a client like `manual_control.py` and I want to set its start and end destinations so that it takes the same route each time Autopilot is enabled, I only need to make use of the scripts in the `agent` folder, and in particular the Behaviour agent? I believe this is the default one used in these Python examples. I came across a flag that can be used to choose from basic, behaviour and roaming_agent.

Thanking you

jackbart94, 2021-02-12:
To use the Autopilot, which again, it's called Traffic Manager, you need to change the Traffic Manager API, which isn't currently using a fixed path, but it is updating it slowly as it goes, since it's meant to roam around the Town without any goal. So it is not a viable choice for you.

The viable choice is using ``automatic_control.py`` and one if its agent, whether Behavior or Roaming (I wouldn't use Basic) but they will probably need some PID tuning, since it's outdated. Everything you need to know about them it's in the ``automatic_control.py`` script, and in the ``PythonAPI/carla/agents/navigation`` folder.

jimmyw404, 2021-02-12:
> Thanks for replying!
> Yes, I have been referring to that page. My main goal is to set the start and end destinations for the ego vehicle so that it follows the same route each time I set it to Autopilot.
We're currently using python scripts to handle this. It would be highly valuable to have directed waypoints for vehicles managed by the Traffic Manager, so if you do develop this please let me know!

@jackbart94

ym2509, 2021-02-14:
Thanks @jackbart94 

@jimmyw404 I might be mistaken but when looking through the scenario_runner files, I came across some XML files in `scenario_runner/srunner/data/` that have a list of waypoints for each route and Town. I was initially hoping to create a similar code that accesses a .XML file containing a list of waypoints for the ego vehicle to follow when Autopilot is enabled. Perhaps the Contributors could comment on that? I will follow @jackbart94's suggestion for making the ego vehicle follow the same route each time the client is run.

ym2509, 2021-04-23:
> To use the Autopilot, which again, is called Traffic Manager, you need to change the Traffic Manager API, which isn't currently using a fixed path, but it is updating it slowly as it goes, since it's meant to roam around the Town without any goal. So it is not a viable choice for you, unless you change it.
> 
> The viable choice is using `automatic_control.py` and one if its agent, whether Behavior or Roaming (I wouldn't use Basic) but they will probably need some PID tuning, since it's outdated. Everything you need to know about them it's in the `automatic_control.py` script, and in the `PythonAPI/carla/agents/navigation` folder.

Hi @jackbart94 
Could you please tell me where in the manual_control.py client is the code located that drives the car to the end destination when autopilot is enabled? Is it using the behaviour or roaming agent modules?

jackbart94, 2021-04-26:
The script ``manual_control.py`` is using the Traffic Manager, not the agent modules.

ymurzello, 2021-04-27:
> The script `manual_control.py` is using the Traffic Manager, not the agent modules.

Thanks for your reply
There doesn't seem to be a TrafficManager.cpp in the [compiled version 0.9.9](https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.9.tar.gz).
Is it possible to modify `manual_control.py` so that when Autopilot is enabled it works with the Behavior Agent and I can set an end destination for the vehicle to drive to?

closed 2021-06-24 by ymurzello

Micahmichael03, 2024-08-09:
> Thanks for your reply There doesn't seem to be a TrafficManager.cpp in the [compiled version 0.9.9](https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/CARLA_0.9.9.tar.gz). Is it possible to [...]

please have you fix your goal