carla-simulator/carla #3860: Spawned vehicles disappear

FiodarKazhamiaka, 2021-02-05:
Version: 0.9.11
Platform: Ubuntu 18.04

Setting: I spawn something like 20 vehicles using the PythonAPI/examples/spawn_npc.py script,
and set up another script that that polls the locations of all actors in the simulation and writes them to a file

Issue: Over the course of a week-long simulation, some actors seem to be disappearing and random points (their location isn't being written to file anymore, so I'm assuming they are not present in world snapshot).

Is this a known bug? Any way to avoid it?

Thanks

jackbart94, 2021-02-05:
Some cars are disappearing because we have a vehicle elimination policy within Traffic Manager, since in some cases they completely block intersections, thus effectively blocking traffic flow altogether. If you remove this, it might happen that vehicles end up stuck for you week-long simulation. I can guide you through removing this feature:

in ```ALSM.cpp```, comment the following lines like this:

``  // Destroy registered vehicle if stuck at a location for too long.
  // if (IsVehicleStuck(max_idle_time.first)
  //     && (current_timestamp.elapsed_seconds - elapsed_last_actor_destruction) > DELTA_TIME_BETWEEN_DESTRUCTIONS) {
  //   registered_vehicles.Destroy(max_idle_time.first);
  //   RemoveActor(max_idle_time.first, true);
  //   elapsed_last_actor_destruction = current_timestamp.elapsed_seconds;
  // }
``

Let me know if this fixes your issue.

FiodarKazhamiaka, 2021-02-16:
@jackbart94 thanks for the explanation! I think it makes sense, but is there away to detect that this occurred using the client Python API?
I tried using the following to detect, and it didn't work: 
`if world.get_actor(id) is None: print("Stuck vehicle eliminated")`

jackbart94, 2021-03-02:
Thanks to your feedback we found an issue with the ``is_alive`` variable. We are working on a fix and it should be merged soon. After it's merged, you can check if a vehicle has been destroyed by Traffic Manager simply using:
```if world.get_actor(id).is_alive is False: print("Vehicle has been eliminated")```

closed 2021-11-18 by jackbart94

sk9073, 2022-10-19:
@jackbart94 I just began using Carla. So I am not sure if I am doing things right. I edited the ALSM.cpp file but the TrafficManager still kept removing dormant vehicles.  Would Carla automatically build those CPP files again every time I launch it or is there some way for me to build those CPP files?

SExpert12, 2025-04-08:
hi, does everyone able to solve this problem. I am also facing this problem.

PR #3940 (2021-03-02): Fix is_alive attribute on actors
#### Description

The actors' attribute is_alive was not properly initialized and considered that all actors were alive by default creating an inconsistency if we used intermediate copies of the actor, e.g. using iterators.

### Fixes:
#3860 

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3
  * **Unreal Engine version(s):** 4.26

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3940)