carla-simulator/carla #3864: Retrieve data (LiDAR, camera, radar) when replaying

xinshuoweng, 2021-02-06:
Hi Guys, 

I am aware of the recording function provided by Carla, which can replay a sequence of data I recorded before. I am wondering if it is possible to also retrieve data (LiDAR, camera, radar) at every timestamp during replaying, instead of just visualizing it? Basically, my goal is to reproduce the exact same sequence of data (all actors, environment, and collected sensor data are the same). 

Thanks!

bernatx, 2021-02-12:
Hi, we have detected a problem during replaying the simulation with sensors, where they are not created properly. We are fixing this, so then, once fixed, when you start replaying you need to find the sensors you want to listen from. They will be created by the replay system, but those sensors are not sending data because nobody is listening on them. That will be your responsability.

xinshuoweng, 2021-02-13:
Thanks a lot for the follow-up! 

Just to confirm, as the bugs are being fixed, should I expect to have this feature in the next version of Carla? 
Also, it would be great if some example code can be added in the documentation for sensor data listening during replay. So far it is unclear to me how to listen to the sensor data in a synchronized mode during a replay. The documentation only provides some simple functions such as start_recorder(), stop_recorder(), show_recorder_file_info()

closed 2021-07-21 by stale[bot]

Morphlng, 2023-05-25:
Hello! Have you solved the problem of listening to the sensor during replay? I found that RGB sensor seems to be working properly, but detectors are not. I've opened a new [issue](https://github.com/carla-simulator/carla/issues/6444) but no one has answered yet. I wonder if you have made any progress?

PR #3923 (2021-02-24): Fixed spawn of sensors during replayer.
#### Description

Fixes the replayer not creating sensors. Now they all spawn as they should.

Fixes #3864

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 3
  * **Unreal Engine version(s):** 4.24

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3923)