carla-simulator/carla #3865: Black Artifacts

EsquilliDev, 2021-02-06:
System Information:

- OS: Ubuntu 20.04
- GPU: RTX 3080 FE (nvidia-driver-460)
- CARLA: 0.9.10.1 (pre-packaged and build from source)
- Vulkan: 1.2.131

Problem details:

- Black artifacts as shown in the pictures below.
- All sensors get affected by it.
- Vulkan libraries are installed.
- The issue happens in both the pre-packaged version and the build from source.

Screenshots:
[image]
[image]
[image]

The only "fix" I have found so far is by using the -opengl4 flag.

qhaas, 2021-02-09:
Do these artifacts also appear in imagery saved to disk, e.g. [like in the tutorial example](https://github.com/carla-simulator/carla/blob/0.9.10.1/PythonAPI/examples/tutorial.py)?

syveqc, 2021-02-09:
I can confirm this issue with vulkan (although to a bit lesser degree on my machine) and that it is present in imagery saved to disk, I attached an example of an RGB-Camera saving to disk using cv2. In my experience it greatly varies when restarting carla, from really annoying to barely noticable, but the artifacts are always there.

https://user-images.githubusercontent.com/16987118/107383887-bc13aa00-6af1-11eb-8e8a-d8fc36a05a63.mp4

.

EsquilliDev, 2021-02-09:
@qhaas As pointed by @syveqc the artifacts also appear in the images saved to disk.

Axel1092, 2021-02-11:
Hi all, this is an issue that happens with some of the newer gpus (such as rtx 3080) when using vulkan in Unreal Engine 4.24. This is an unreal-driver issue that seems to be fixed in the latest version (4.26) and we plan to switch to that version for the next release.

qhaas, 2021-02-11:
> Hi all, this is an issue that happens with some of the newer gpus (such as rtx 3080) when using vulkan in Unreal Engine 4.24. This is an unreal-driver issue that seems to be fixed in the latest version (4.26) and we plan to switch to that version for the next release.

That is good to hear, looking forward to official headless/displayless vulkan (Issue #3800), fixes to this issue's artifacts, and the OpenGL artifacts of Issue #3377.

On the chance that an upgrade to UE 4.25.x fixes the OP's issue, we got CARLA 0.9.11 (unofficially) working with UE 4.25.4 over in [Issue #3214](https://github.com/carla-simulator/carla/issues/3214#issuecomment-769140155)

swsamleo, 2022-11-07:
The problem still exist with the lasted version of CARLA (0.9.13) with A100 card

jalane76, 2023-11-01:
I am also having this issue:

- OS: Ubuntu 22.04
- GPU: A100
- Nvidia Driver: 525.85.05
- CARLA: 0.9.14 (Docker container)

The issue occurs with and without the `-opengl` flag.  This image has also been saved to disk.

[image]

Barnonewdm, 2023-12-21:
This problem also happens from our side. A100.

buaazeus, 2024-02-22:
same issue
GPU:A100
carla:0.9.10
Vulkan.
When use opengl, no artifacts.

Blyron, 2024-11-07:
We do not support versions <= 0.9.12 . Please try this issue in 0.9.14 and later versions. OpenGL is not supported neither

closed 2025-05-08 by Blyron