carla-simulator/carla #3866: Is it possible to run UE4 and CARLA in AWS virtual machine?

zhiranyan-edmrc, 2021-02-06:
Hello,

I am using AWS p2.xlarge (Ubuntu 18.04) and want to install UE4 in the instance in order to run CARLA.

But I cannot run the UE4. I installed it according to this tutorial (https://carla.readthedocs.io/en/latest/build_linux/).
Always get the error: cannot find a compatible vulkan driver or device.

So is it possible torun UE4 and CARLA in AWS virtual machine?

Could anyone give me some tipps? Thank you very much!

qhaas, 2021-02-08:
The client is CPU only, but the server (i.e. the part that needs UnrealEngine) needs a GPU capable of running OpenGL or Vulkan attached to the VM, doubt CPU only OpenGL will be usable.  I have not ran it on AWS, where [GPUs appear to be an option](https://aws.amazon.com/nvidia), but have ran it on our internal cloud which is backed by [openstack](https://www.openstack.org/).

Typically, you are not running an x-server on a cloud VM, so you will be limited to headless with no x-server opengl runs.  [vulkan doesn't support x-server-less headless](https://github.com/carla-simulator/carla/issues/3800#issuecomment-768993043) without an upgrade to UE 4.25.  You can use OpenGL headless without an x-server out-of-the-box, but [there are artifacts in certain weather conditions](https://github.com/carla-simulator/carla/issues/3377#issuecomment-766937623).

zhiranyan-edmrc, 2021-02-09:
> 
> 
> The client is CPU only, but the server (i.e. the part that needs UnrealEngine) needs a GPU capable of running OpenGL or Vulkan attached to the VM, doubt CPU only OpenGL will be usable. I have not ran it on AWS, where [GPUs appear to be an option](https://aws.amazon.com/nvidia), but have ran it on my organization's internal cloud which is backed by [openstack](https://www.openstack.org/).
> 
> Typically, you are not running an x-server on a cloud VM, so you will be limited to headless with no x-server opengl runs. [vulkan doesn't support x-server-less headless](https://github.com/carla-simulator/carla/issues/3800#issuecomment-768993043) without an upgrade to UE 4.25. You can use OpenGL headless without an x-server out-of-the-box, but [there are artifacts in certain weather conditions](https://github.com/carla-simulator/carla/issues/3377#issuecomment-766937623).
> 
> To use headless opengl, [take a look at their docker example](https://github.com/carla-simulator/carla/blob/0.9.11/Util/Docker/Release.Dockerfile#L37), we start it like `SDL_VIDEODRIVER=offscreen CarlaUE4.sh -opengl`

Hi qhaas, thanks for your reply. I could give you more information about me. Now, I am running with a GPU-capable AWS virtual machine (e.g. p2.xlarge, GPU is NVIDIA Tesla K80). I try to open UE4.24 but I get the error "Cannot find a compatible Vulkan device or driver. Try updating your video card to a more recent version and make sure your video card supports Vulkan." that is the same like this issue (https://answers.unrealengine.com/questions/883700/ubuntu-1804-cannot-find-a-compatible-vulkan-device.html).  Ihave tried to method it mentioned. But I still coundn't open UE4.

So, I start to consider if it is a issue about NVIDIA driver. I typed "nvidia-smi" and get the correct information. However, when I go to "system settings->details->about", it shows "graphics: llvmpipe (LLVM10.0.0, 256 bits)" which means that NVIDIA driver is not being used. But I am sure I selected NVIDIA driver, because I can train my neutral network with this virtual machine with GPU.

I have tried your suggestion to open UE4 with the command "cd ~/UnrealEngine_4.24/Engine/Binaries/Linux && SDL_VIDEODRIVER=offscreen ./UE4Editor -opengl ". But it does not work for me.

qhaas, 2021-02-09:
My above snippet was for running CARLA (not the UnrealEditor) from the command-line without a GUI.

If you want to use UnrealEditor with a GUI via x-forwarding or VNC (or any other 3D accelerated software where the rendering is done remotely), you [cannot use vulkan](https://github.com/VirtualGL/virtualgl/issues/37) and will need [virtualgl](https://virtualgl.org/About/Introduction).

Try to get remote rendering with OpenGL and a display working with something simpler than UnrealEditor first, like [glxgears](https://wiki.archlinux.org/index.php/benchmarking#glxgears), which is included in some distros' repositories.  If glxgears won't work, UnrealEditor won't either.

cleemesser, 2021-06-27:
I just tried it, and UnrealEngine5 (early access, built from source on linux) no longer supports OpenGL but instead requires vulkan :-(

closed 2023-06-02 by stale[bot]