carla-simulator/carla #3867: world.apply_settings() slow in 0.9.11 synchronous mode

syveqc, 2021-02-08:
Hi,

I just upgraded to Carla 0.9.11 and now my code runs slow. (Ubuntu 20.04). After some digging I found that world.apply_settings() needs significantly more time than before. 
Code example:
```
import carla
import time

#load carla and apply basic settings
client = carla.Client("localhost", 2000)
world = client.load_world("Town04")
settings = world.get_settings()
settings.no_rendering_mode = True
settings.fixed_delta_seconds = 0.1
settings.synchronous_mode = True
world.apply_settings(settings)

#needed because first tick takes longer than subsequent ones sometimes
world.tick(1)

#tick without applying settings
start = time.time()
world.tick(1)
print(f"normal step needed {int((time.time()-start)*1000)}ms")

#tick after applying settings
start = time.time()
world.apply_settings(settings)
print(f"apply settings needed {int((time.time()-start)*1000)}ms")
world.tick(1)
print(f"apply settings and step needed {int((time.time()-start)*1000)}ms")
```

I tested it in opengl and vulkan mode, the error exists in both. (in vulkan the non-apply_settings() steps are a lot quicker, the apply_settings() steps need nearly identical time) 

Output in 0.9.10 on my machine (time may vary a couple ms)
```
normal step needed 6ms
apply settings needed 0ms
apply settings and step needed 9ms
```
Output in 0.9.11 on my machine (time may vary a couple ms)
```
normal step needed 8ms
apply settings needed 1002ms
apply settings and step needed 1003ms
```

For completion's sake here is my hardware, but the behavior was also confirmed on another machine, so I do not think the hardware actually matters:
- CPU: AMD Ryzen 5950x
- GPU: Nvidia RTX 3090
- RAM: 32GB 3600MHz C16

Thanks in advance for your help!

hh0rva1h, 2021-02-09:
I can confirm this.
@germanros1987 could you please triage this issue?

corkyw10, 2021-03-12:
@bernatx could you follow up on this please?

hh0rva1h, 2021-08-19:
go away stale bot

hh0rva1h, 2022-02-16:
@bernatx Please follow up, Code example is given, should be quick to look into.

hh0rva1h, 2022-04-29:
@bernatx Still an issue with 0.9.13, why is it so much slower? This is actually relevant for people doing reinforcement learning and applying settings after each episode: 1 second wait time after each episode can amount to much training time being wasted, especially at the begin of the training when episodes are very short!

closed 2026-11-13 by carla-triage[bot]