carla-simulator/carla #3871: GNSS to XYZ

SebastianMalert, 2021-02-09:
I found a simple way to transform the latitude and longitude GNSS data into the XYZ coordinatesystem.I don't understand why it works, but I would like to. The basic idea is, that it is just a transformation from a spheric to a cartesian coordinate system. My solution is as followed:
`    def get_gnss_data(self, raw_data):
        timestamp = raw_data.timestamp

        lat_rad = (np.deg2rad(raw_data.latitude) + np.pi) % (2 * np.pi) - np.pi
        lon_rad = (np.deg2rad(raw_data.longitude) + np.pi) % (2 * np.pi) - np.pi
        R = 6378135 # Aequatorradii

        x = R * np.sin(lon_rad) * np.cos(lat_rad)       # iO
        y = R * np.sin(-lat_rad)                        # iO
        z = raw_data.altitude

        self._gnss_data = [timestamp, x, y, z]
`

This functin is part from an object function and is called by a lambda function everytime, when the gnss sensor receives new data. 
The Question is: Why does it work? 

Usually it should something like this:
` x = R * np.sin(lon_rad) * np.cos(lat_rad)
  y = R * np.sin(lon_rad) * np.sin(lat_rad)
  z = R * np.cos(lon_rad)
`

corkyw10, 2021-03-12:
@DSantosO could you have a look at this please?

tiger-bug, 2021-08-18:
I would be interested to know this as well if the issue can be reopened.