carla-simulator/carla #3873: Does Carla in a Docker support video streaming?

Michal-Kolomanski, 2021-02-09:
System specification:
- server: Ubuntu 18.04.4 LTS
- Carla 0.9.5
- Python 3.7

The script which runs a container:
[image]

Let's look closely to **manual_control.py** example:

1.  Version with a local server:
[image]
It works perfectly fine. 

2. Version with the server:
[image]

As we can see, there is not rendering at all. 
I wonder if there is a possibility to get video from Carla Docker which is on the server? 

When I was searching for a solution, I found: [carlaviz](https://carla.readthedocs.io/en/latest/plugins_carlaviz/)
May it solve the problem?

WoLewicki, 2021-02-15:
I have the same problem. Connecting to the Carla server in docker image resolves in being able to connect the client to the server, but it seems like the server itself is not generating frames of simulation. Can it have something to do with the `pygame` configuration in `manual_control.py`? Do you use docker images to test the visual side of the project @marcgpuig @bernatx ? If not, maybe you could highlight some options/paths we can check to try and make it work?

Michal-Kolomanski, 2021-02-21:
When I run a container without `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0`
[image]
Results are the same. 

I also tried:
[image]
but results are still the same - 0 FPS from the server and black screen with some params.

Furthermore, the problem is only with graphical display because the connection is stable.
A sample test:
```
import glob
import os
import sys
import time
import carla

client = carla.Client('127.0.0.1', 2000)
# client.set_timeout(2.0) # seconds
start = time.time()
try:
    world = client.get_world()
    print("OK1")
except:
    world = client.get_world()
    print("OK2")

stop = time.time()
print(stop - start)

```
The output:
[image]
Surprisingly, the first connection is always unsuccessful and the second one is always successful. I found it strange but I can live with that. 

My question is how can I run Carla in a Docker and be able to see the graphical display for example in `manual_control.py`?

psarka, 2021-03-04:
I have the same issue, the sample test passes with `OK1`, but I get exactly the same black screen. The controls are also not responsive, or at least the throtle, steer and break indicators do not change when I press the corresponding keys.

psarka, 2021-03-04:
Surprisingly, for me rendering works with carla 0.9.10 @Michal-Kolomanski, could you give it a try? It does work with 0.9.11.

WoLewicki, 2021-03-04:
@psarka can you provide the commands you are running the carla docker with?

psarka, 2021-03-04:
Sure, with

```
docker run --rm --runtime=nvidia -d -p 2000-2002:2000-2002 -e SDL_VIDEODRIVER=offscreen carlasim/carla:0.9.10 bash -c "./CarlaUE4.sh -opengl -nosound -carla-server"

```

Michal-Kolomanski, 2021-03-04:
> Sure, with
> 
> ```
> docker run --rm --runtime=nvidia -d -p 2000-2002:2000-2002 -e SDL_VIDEODRIVER=offscreen carlasim/carla:0.9.10 bash -c "./CarlaUE4.sh -opengl -nosound -carla-server"
> ```

Thanks for the response @psarka. I pulled `carlasim/carla:0.9.10` on the Ubuntu server and downloaded `Carla 0.9.10` locally on Windows 10.  Furthermore, I have used your script to run a Docker but I still got the black screen in `manual_control.py` from Carla 0.9.10.

Could you please say where do you run your Docker container? It's locally on Windows or maybe remotely on Ubuntu 18.04.4 LTS?

Moreover, I am quite confused with `SDL_VIDEODRIVER=offscreen` parameter. I thought that this parameter is used to NOT generate any graphic. As we care about generating the graphic, I have encountered the usage of `-e DISPLAY=$DISPLAY` parameter. But for me `$DISPLAY` variable is empty. I don't know what this variable should be. 

Moreover, when I use  `-e DISPLAY=$DISPLAY` and run `script.sh`: 
`docker run --runtime=nvidia -p 2000-2002:2000-2002 -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix carlasim/carla:0.9.10 bash -c "./CarlaUE4.sh -opengl -nosound -carla-server"`

I got: 
[image]

So it simply exits after 1s.

psarka, 2021-03-04:
Thanks for checking, it is sad that 0.9.10 does not work for you.

I'm running the server container on ubuntu 20.04.1, and I'm running the client on a different ubuntu 20.04.2. By the way, my client is 0.9.11, which I installed with apt-get install (0.9.10 is not available, even though the docs say that it should be. But 9.11 client works with 9.10 server so not an issue at least for the moment).

SDL_VIDEODRIVER=offscreen to my understanding means to render graphics to some fake display (but still render!).

The command that you wrote shouldn't work. You don't have a display (thats why DISPLAY is empty), and you tell carla to render to it, and it crashes. SDL_VIDEODRIVER=offscreen is essential.

Michal-Kolomanski, 2021-03-04:
> Thanks for checking, it is sad that 0.9.10 does not work for you.
>
> I'm running the server container on ubuntu 20.04.1, and I'm running the client on a different ubuntu 20.04.2. By the way, my client [...]

Thank you @psarka, this is a very helpful tip for me. So probably there is a problem with Windows -> Ubuntu connection. 
I will try connecting from some virtual machine with an Ubuntu system. 

Could you say something about your performance?  Are responsiveness and FPS satisfactory?

psarka, 2021-03-04:
No, not satisfactory at all, around 10 FPS. I tried to compile the vulkan version, but for that one I get the black screen :(

corkyw10, 2021-03-12:
@bernatx could you have a look at this please?

qhaas, 2021-04-02:
> SDL_VIDEODRIVER=offscreen to my understanding means to render graphics to some fake display (but still render!).

We explored the extensions Epic has made to SDL [in Issue #225](https://github.com/carla-simulator/carla/issues/225#issuecomment-507666953).  Epic [forked SDL and bundled it with UE](https://github.com/EpicGames/UnrealEngine/tree/4.24.3-release/Engine/Source/ThirdParty/SDL2), some changes [have been upstreamed](https://github.com/libsdl-org/SDL/commit/68985371a071b97b961df56d040fbb5ca6493cf3).

There is some overloaded terminology at play here.  Headless/offscreen can mean one of three things:
1. A CARLA GUI being displayed on a host system's physically attached display while being executed inside a container (vulkan/opengl)
2. A CARLA GUI being displayed on a remote system (OpenGL + VirtualGL)
3. No GUI and no X-server, the images/data are just dumped to flat files (OpenGL and [modified CARLA 0.9.11 builds using UE 4.25.x + vulkan](https://github.com/carla-simulator/carla/issues/3214#issuecomment-769140155))

closed 2023-06-02 by stale[bot]