carla-simulator/carla #3876: Spawn Vehicle near Ego

Stefano38, 2021-02-10:
Hi all, I'm working on CARLA and scenario runner. 
System specification:
-OS Windows 10
-Carla 0.9.11
-Python 3.7
I was wondering if there is any way to populate the world only near the EGO vehicle or near a coordinate. In other words, spawn a certain number of vehicles within a radius, centred in EGO vehicle or waypoint.
Thanks in advance.

glopezdiest, 2021-02-11:
Hey @Stefano38. There is no straightforward way to do so, so you'd have to create one. Some brainstorming here:

The usual way to spawn vehicles is to make use of the spawn points but those will result in limited vehicles being spawned so you might want to make use of the Waypoint API. You could do something like get a random point around the ego and then get do `carla.Map.get_waypoint()` to get the closest center lane (and spawn a vehicle there).

You can also use the WaypointAPI to create vehicles at a left / right lane, in front / behind..., but that might be harder to automatize.

However, which is your overall goal? Do you want your spawned vehicles to have an autopilot? If so, very soon the surrounding of your ego vehicle will be empty again.

Anyhow, let me know if that helps.

Stefano38, 2021-02-11:
Hi @glopezdiest  thanks for your fast answer, I give you a little more contest. I'm creating traffic scenarios with Openscenario configuration and now I want to simulate with a more realistic situation, adding some vehicle with AUTOPILOT surrounding my ego vehicle. I can't populate all the maps because of my low power PC. So my favourite solution would be to spawn 5 to 10 vehicles near EGO, complete my scenario, destroy all actor and restart.
Thanks in advance.

glopezdiest, 2021-02-11:
>  I can't populate all the maps because of my low power PC

Then, I'd suggest running the autopilot with the hybrid mode on, which cause far away vehicles to move without physics (check `set_hybrid_physics_mode` at the [TrafficManager](https://carla.readthedocs.io/en/latest/python_api/#carlatrafficmanager) and set its radius to your adequate value.

If that still doesn't work for you, then my previous answer still stands, as there is no current way to implement the spawning you are trying.

Zhuweilong123, 2021-02-20:
@Stefano38
I write a simple code, you can try it in your  PC.

````
import carla
import numpy as np
import math

def spawn_vehicles_around_ego_vehicles(ego_vehicle, radius, spawn_points, numbers_of_vehicles):
    # parameters:
    # ego_vehicle :: your target vehicle
    # radius :: the distance limitation between ego-vehicle and other free-vehicles
    # spawn_points  :: the available spawn points in current map
    # numbers_of_vehicles :: the number of free-vehicles around ego-vehicle that you need
    np.random.shuffle(spawn_points)  # shuffle  all the spawn points
    ego_location = ego_vehicle.get_location()
    accessible_points = []
    for spawn_point in spawn_points:
        dis = math.sqrt((ego_location.x-spawn_point.location.x)**2 + (ego_location.y-spawn_point.location.y)**2)
        # it also can include z-coordinate,but it is unnecessary
        if dis < radius:
            print(dis)
            accessible_points.append(spawn_point)

    vehicle_bps = world.get_blueprint_library().filter('vehicle.*.*')   # don't specify the type of vehicle
    vehicle_bps = [x for x in vehicle_bps if int(x.get_attribute('number_of_wheels')) == 4]  # only choose car with 4 wheels

    vehicle_list = []  # keep the spawned vehicle in vehicle_list, because we need to link them with traffic_manager
    if len(accessible_points) < numbers_of_vehicles:
        # if your radius is relatively small,the satisfied points may be insufficient
        numbers_of_vehicles = len(accessible_points)

    for i in range(numbers_of_vehicles):  # generate the free vehicle
        point = accessible_points[i]
        vehicle_bp = np.random.choice(vehicle_bps)
        try:
            vehicle = world.spawn_actor(vehicle_bp, point)
            vehicle_list.append(vehicle)
        except:
            print('failed')  # if failed, print the hints.
            pass
# you also can add those free vehicle into trafficemanager,and set them to autopilot.
# Only need to get rid of comments for below code. Otherwise, the those vehicle will be static
    # tm = client.get_trafficmanager()  # create a TM object
    # tm.global_percentage_speed_difference(10.0)  # set the global speed limitation
    # tm_port = tm.get_port()  # get the port of tm. we need add vehicle to tm by this port
    # for v in vehicle_list:  # set every vehicle's mode
    #     v.set_autopilot(True, tm_port)  # you can get those functions detail in carla document
    #     tm.ignore_lights_percentage(v, 0)
    #     tm.distance_to_leading_vehicle(v, 0.5)
    #     tm.vehicle_percentage_speed_difference(v, -20)
# create client object
client = carla.Client("localhost", 2000)
client.set_timeout(10)

# create world object
world = client.load_world('Town05')
spawn_points = world.get_map().get_spawn_points()
model3_bp = world.get_blueprint_library().find('vehicle.tesla.model3')
model3_bp.set_attribute('color', '255,0,0')   # set vehicle's color to red
model3_spawn_point = np.random.choice(spawn_points)
model3_actor = world.spawn_actor(model3_bp, model3_spawn_point)
spawn_vehicles_around_ego_vehicles(ego_vehicle=model3_actor, radius=100,
                                   spawn_points=spawn_points, numbers_of_vehicles=10)
 # model3_actor.set_autopilot(True)  # set the ego-vehicle to autopilot
```

Stefano38, 2021-02-24:
Thank you @Zhuweilong123 for your answer, we solved the problem with a code very similar to yours.

closed 2021-03-12 by corkyw10

jhih-ching-yeh, 2022-06-20:
Hi, I ran into another problem when I run this code in deep reinforcement learning program.
My program showed this error message "terminate called after throwing an instance of 'clmdep_msgpack::v1::type_error' what(): std::bad_cast" after ran 100 times.
Could you help me? Thank you!!