carla-simulator/carla #3877: Light rendering is weird, and Lane direction is incorrect from the map that I export from RoadRunner (CARLA 0.9.11)

siraphopp, 2021-02-11:
I draw a map in RoadRunner with setting left-hand traffic rule

, and I did "Map ingestion in a build from source" method from
[https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-build-from-source](url)

Here is the result.

[image]

The light is weird. I encountered this issue once when I used CARLA version 0.9.10, but I followed this issue #3426 . It could fix that, but cannot fix with CARLA version 0.9.11

, and when I spawned a car with manual_control.py, and toggled autopilot. The car was driven in right lane, but I set left-hand traffic rule when I export map from RoadRunner

, and RoutePlanner lines were not connected at every junction.

Madecu, 2021-02-12:
Hi @siraphopp, 

These shadows are caused by Mesh Distance Fields. We believe that because of having the road and lane marking separately, unreal engine makes a mess when calculating those shadows.

To remove them, you have to go to each mesh of road and lane marking and set the distance fields resolution scale to 0.0.

[image]

siraphopp, 2021-02-16:
Hi @Madecu and thanks for your reply.

[image]

The issue was fixed.

closed 2021-02-19 by XGodina