carla-simulator/carla #3879: make package error in carla-0.9.11

whuxhzhang, 2021-02-11:
ubuntu18.04 
carla-0.9.11

I download the source code of carla-0.9.11 (master branch) and compile it,  but always meet the following error：
rsync: change_dir "/home/zhangxinhong/carla0.9.11//./Plugins" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]

How can I fix it ?

SergioBarroso, 2021-02-11:
Hi, I solved this error using make plugins

qhaas, 2021-02-11:
Haven't seen this problem in particular, but I've been using newer versions of cmake than what is in Ubuntu 18.04 for a while to compile CARLA.  I've [noticed issues](https://answers.launchpad.net/ubuntu/+source/cmake/+question/692682) when compiling recent software with the old version of cmake that is included in Ubuntu 18.04.  Thankfully, the vendor [provides the latest version in their Ubuntu repo](https://blog.kitware.com/ubuntu-cmake-repository-now-available/).

thillRobot, 2021-02-11:
Have you looked at #3758? This sounds like a similar issue.

whuxhzhang, 2021-02-12:
> Hi, I solved this error using make plugins

ummm，I don't know how to making plugins, Can you tell me how to solve it ？

whuxhzhang, 2021-02-13:
> Hi, I solved this error using make plugins

Thank you,  I solved it ,  too.

whuxhzhang, 2021-02-13:
> Have you looked at #3758? This sounds like a similar issue.

Thank you very much. It did work for me

closed 2021-03-15 by corkyw10