carla-simulator/carla #3881: manual_control.py example

Fady-Maged-Louis, 2021-02-11:
**Describe our question or idea**
I run manual_control.py example.

**Expected behavior**
pygame window flashes then close and error appear

```
 File "manual_control.py", line 1187, in <module>
    main()
  File "manual_control.py", line 1179, in main
    game_loop(args)
  File "manual_control.py", line 1095, in game_loop
    world = World(client.get_world(), hud, args)
RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:10000
```

**Desktop 
 - OS: windows10
 - CARLA Version :CARLA 0.9.10
 - Python version 3.7.9
[image]

glopezdiest, 2021-02-11:
Just saw that you are using port 10000, which isn't the default value. Are you doing this in both `spawn_npc.py` and `manual_control.py`? And why aren't you using the default one?

Fady-Maged-Louis, 2021-02-11:
spawn_npc.py run using port 2000 
manual_control.py run using port 10000 I try to change it to 2000 but nothing change

jackbart94, 2021-02-12:
Add ``--port 2000`` to whatever you are executing in the command line, and see if the error is still happening. Please send a screenshot of whatever it outputs, if it's still not working, because the default value is definitely 2000.

Fady-Maged-Louis, 2021-02-12:
[image]
[image]
I try to add `--port 2000` but the error still exist

closed 2021-03-11 by Fady-Maged-Louis