carla-simulator/carla #3882: Downgrade Carla 9.11 -> 9.10

LuigiRu, 2021-02-11:
I am using Ubuntu 18.04 and install Carla using the got hub rep. I accidentally installed the wrong version of Carla (9.11) and now my Ros bridge does not work anymore.
Is there a quick way to install Carla 9.10, without problems? Is there a guide? 

Thanks

jackbart94, 2021-02-12:
If you are using the GitHub repository, from inside the folder where you cloned it, you can just do:
1. ``git checkout 0.9.10``
2. ``make clean LibCarla PythonAPI``
and you have CARLA 0.9.10.

LuigiRu, 2021-02-12:
Hi Jacopo and thanks for your reply. By doing so I am able to install Carla 0.9.10 however the installation is "dirty" and my branch is HEAD. Moreover by default I get only the carla0.9.10-python3.8egg, and the 2.7 egg is not installed. 
Even if I export it manually in the bashrc file I get errors for "no module named Carla", do you have any suggestions?

jackbart94, 2021-02-15:
> Hi @jackbart94 ! and thanks for your reply.
> By doing as you suggested I am able to install Carla 0.9.10, however the installation is "carla-0.9.10-dirty" and my branch is HEAD.
If it's not causing you any particular issue, then don't worry.
> Moreover when I launch `make clean LibCarla PythonAPI` I get only the carla0.9.10-python3.6.egg in` PythonAPI/carla/dist`, and the 2.7 egg is not added.
> Since I need python2 also for the carla-ros-bridge I have launched : `sudo make PythonAPI ARGS="-python-version=2.7. ` and manually exported the `PYTHONPATH` of the 2.7 egg in the .bashrc file.
It would have been enough to do ``make PythonAPI.2``.
> By doing so I have the 2 egg files in the ` PythonAPI/carla/dist`, however every time that I try to open a new terminal I get the following error, which is generated by the command in the bashrc file: `python3 -c import carla; print("Success")` :
> 
> ```
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "/home/username/carla/PythonAPI/carla/dist/carla-0.9.10-py2.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
>   File "/home/username/carla/PythonAPI/carla/dist/carla-0.9.10-py2.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
>   File "/home/username/carla/PythonAPI/carla/dist/carla-0.9.10-py2.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
>   File "/usr/lib/python3.6/imp.py", line 343, in load_dynamic
>     return _load(spec)
> ImportError: /home/username/.cache/Python-Eggs/carla-0.9.10-py2.7-linux-x86_64.egg-tmp/carla/libcarla.so: undefined symbol: PyClass_Type
> ```
> 
> My python2 version is 2.7 and python3 is 3.6.9.
> Do you have any suggestion on how to fix this? What am I doing wrong?

Manually delete the python3 .egg file and don't use ``python3`` but just ``python`` or it won't work with python2.

closed 2021-02-17 by LuigiRu

guofan999, 2021-11-14:
ImportError: /home/username/.cache/Python-Eggs/carla-0.9.10-py2.7-linux-x86_64.egg-tmp/carla/libcarla.so: undefined symbol: PyClass_Type
I have the same problem, how did you fix it? please.