carla-simulator/carla #3887: Carla / UE install

ipirva, 2021-02-14:
Hello,

I am trying to install Carla on Linux ARMv8 (Jetson Xavier NX) using the Linux installation guide: https://carla.readthedocs.io/en/latest/build_linux/

```
git clone -b release https://github.com/EpicGames/UnrealEngine.git
./Setup.sh
Updating dependencies:   0% (0/63142)...
Failed to download 'http://cdn.unrealengine.com/dependencies/UnrealEngine-9954827-b7c45323c9f34d9abe0bddc50d4c769c/01c0d68d8182590bd9ea2b82a37745c39e5bbd13': The type initializer for 'System.Net.WebRequest' threw an exception. (TypeInitializationException)
Result: 1
```

I have tried with the UE4.24 release as exactly the example from the installation guide, but I ran into the same error.

I can wget the file:
`
 wget -c http://cdn.unrealengine.com/dependencies/UnrealEngine-9954827-b7c45323c9f34d9abe0bddc50d4c769c/01c0d68d8182590bd9ea2b82a37745c39e5bbd13
--2021-02-14 19:44:13--  http://cdn.unrealengine.com/dependencies/UnrealEngine-9954827-b7c45323c9f34d9abe0bddc50d4c769c/01c0d68d8182590bd9ea2b82a37745c39e5bbd13
Resolving cdn.unrealengine.com (cdn.unrealengine.com)... 13.224.234.98
Connecting to cdn.unrealengine.com (cdn.unrealengine.com)|13.224.234.98|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2645976 (2,5M) [text/plain]
Saving to: ‘01c0d68d8182590bd9ea2b82a37745c39e5bbd13’

01c0d68d8182590bd9ea2b82a37745c39e5bbd13  100%[====================================================================================>]   2,52M  --.-KB/s    in 0,09s   

2021-02-14 19:44:13 (27,5 MB/s) - ‘01c0d68d8182590bd9ea2b82a37745c39e5bbd13’ saved [2645976/2645976]

`

May I ask for help please? Thank you.

qhaas, 2021-02-19:
Are you planning to run the server or just the client on ARM?  If the latter, which I've seen more often, you don't need the UnrealEngine to compile just the client.  You can just run `make PythonAPI`.

I'm not sure if UnrealEngine 4.24 is fully supported on ARM for the purposes of CARLA serve, but I could be wrong (we had trouble with ppc64le).

ipirva, 2021-02-28:
I plan to run both, client and server, on an NVIDIA Jetson Xavier NX and take advantage of the acceleration hardware. I have not been able to setup the UE.

corkyw10, 2021-03-15:
@bernatx could you have a look at this please?

closed 2023-06-02 by stale[bot]

raul-parada, 2024-11-12:
Hi. Someone manage to run CARLA client on ARM?