carla-simulator/carla #3888: Vehicle factory not loading up

parth3153, 2021-02-14:
so i created my vehicle and rigged the bones in blender, did the physics assessment as in documentation, now in the last step of documentation its unclear its written to open the vehicle factory and add the vehicle but in my case the file is not loading up. Please help team Carla and community. 
Thankyou

-->

XGodina, 2021-02-19:
Hi,

I know, that our documentation about how to implement vehicles in CARLA is a bit confusing. Now we are working to update.
This file is in UnrealEngine. Is a Blueprint that you can open and then put your vehicle there.

[image]

[image]

parth3153, 2021-02-22:
no, thats 
[image]
not the problem i am not able to open the ue4 Editor to open the VehicleFactory asset

and i am not able to open it from the engine itself, it ask me to download carsim which is not free and not make sense because its not mentioned in the documentation, idk the reason for all of these

XGodina, 2021-02-22:
I see,

From what I see in the picture.

Estas intentado abrir UE4, en una versión nuestra compilada. If you need to add vehicles you need to download whole project and UE4.

[image]

Here you have the documentation on how to download the project and UE4 with win and Linux

XGodina, 2021-02-22:
https://carla.readthedocs.io/en/latest/#getting-started

closed 2021-07-21 by stale[bot]