carla-simulator/carla #3889: ImportError: No module named util

Johanfanas, 2021-02-14:
Hi,

I built carla the developer mode and I am trying to roslaunch carla_ros_bridge carla_ros_bridge_with_example_ego_vehicle.launch and I am getting "No module name util". When I run one of the example codes in the PythonAPI using python3 there is no issue. Therefore, I know that error is because the egg file for carla 0.9.10 is for python3.7 and my system is probably running it with python2.7, but I don't have the egg file for python2.7.

[image]
. My pc has Ubuntu 18.04 and I am using ros melodic.

Any suggestions?

jackbart94, 2021-02-15:
Just do ``make PythonAPI.2`` and you should have the .egg file for Python 2.7.

Johanfanas, 2021-02-15:
The carla I downloaded from here https://github.com/carla-simulator/carla/releases/tag/0.9.10 doesn't have any make file or anything so I can't generate the egg file

jackbart94, 2021-02-15:
> The carla I downloaded from here https://github.com/carla-simulator/carla/releases/tag/0.9.10 doesn't have any make file or anything so I can't generate the egg file

Please, download the `0.9.10.1` version, or also the `0.9.11` version, which also includes the .egg file for Python 2. The `0.9.10` is the only one that doesn't.

Johanfanas, 2021-02-15:
Thanks, I got it working

closed 2021-02-16 by jackbart94

likith3611, 2022-08-21:
Hii I am facing the same error now, my system is running noetic with supports python3 alone and because of that I had set the egg file to python 3.7 version and am still facing the same issue any help here please?

AOOOOOA, 2023-05-16:
> Thanks, I got it working

Hi, May I ask how did you fix that? Thanks! @Johanfanas