carla-simulator/carla #3890: How to  generate waypoints between two location in carla?

ImtiazUlHassan, 2021-02-15:
Is there a way to generate waypoints  which would connect two points on carla. For example i want to generate waypoints which would connect the following two points in the map ?

[image]

jackbart94, 2021-02-15:
In the Python API I don't think there's a method, but we have everything you need in the `agents` folder, specifically in the following scripts: ``global_route_planner.py`` and ``local_planner.py``. You might also want to take a look at the rest of them.

Zhuweilong123, 2021-02-19:
`
import carla

from agents.navigation.global_route_planner import GlobalRoutePlanner
from agents.navigation.global_route_planner_dao import GlobalRoutePlannerDAO

client = carla.Client("localhost", 2000)
client.set_timeout(10)
world = client.load_world('Town01')
amap = world.get_map()
sampling_resolution = 2
dao = GlobalRoutePlannerDAO(amap, sampling_resolution)
grp = GlobalRoutePlanner(dao)
grp.setup()
spawn_points = world.get_map().get_spawn_points()
a = carla.Location(spawn_points[50].location)
b = carla.Location(spawn_points[100].location)
w1 = grp.trace_route(a, b)  # there are other funcations can be  used  to gener
ate a route in GlobalRoutePlanner.
i = 0
for w in w1:
    if i % 10 == 0:
        world.debug.draw_string(w[0].transform.location, 'O', draw_shadow=False,
                                color=carla.Color(r=255, g=0, b=0), life_time=120.0,
                                persistent_lines=True)
    else:
        world.debug.draw_string(w[0].transform.location, 'O', draw_shadow=False,
                                color = carla.Color(r=0, g=0, b=255), life_time=1000.0,
                                persistent_lines=True)
    i += 1
`
you can test this code, and it can get a waypoints between two locations.
[image]

I also have a question. Can you determine the coordinate of every location? I want to match location's coordinate with its Id. Now, I only can choose  locations from apawn_points.(spawn_points = world.get_map().get_spawn_points()). I won't know its loaction in carla map before I draw it or generate a vehicle in this point in carla.
I want get a match relationship like this picture:
[image]

Zhuweilong123, 2021-02-19:
@jackbart94 @ImtiazUlHassan

Zhuweilong123, 2021-02-19:
@ImtiazUlHassan @jackbart94

ImtiazUlHassan, 2021-02-19:
> @ImtiazUlHassan @jackbart94

`w[0].transform.location`

You can get the coordinate of the way-point using this. you can store it in a variable. you can also get road id lane id way-point id using 

`w[0]
.lane_id` 

`w[0]
.road_id` 

to check more functions you can use `dir(w[0])`

Zhuweilong123, 2021-02-19:
Thanks for your reply. Now I know how to obtain those parameters. I still have a question. Can I mark those information in the carla map. @ImtiazUlHassan

ImtiazUlHassan, 2021-02-19:
> Thanks for your reply. Now I know how to obtain those parameters. I still have a question. Can I mark those information in the carla map. @ImtiazUlHassan

You are welcome 
and yes you can draw/mark them on map. if you want to draw ids than you will have to convert them into strings.

i.e  `mark=str(w[0].lane_id)`

then 

  ` world.debug.draw_string(w[0].transform.location,mark, draw_shadow=False,
color=carla.Color(r=255, g=0, b=0), life_time=120.0,
persistent_lines=True) `

Zhuweilong123, 2021-02-19:
I got it. Thank you very much!
Best wish to you！

Zhuweilong123, 2021-02-19:
I got it. Thank you very much!
Best wish to you！ @ImtiazUlHassan

closed 2021-02-22 by ImtiazUlHassan

ManishwarG, 2022-10-28:
> ` import carla
>
> from agents.navigation.global_route_planner import GlobalRoutePlanner from agents.navigation.global_route_planner_dao import GlobalRoutePlannerDAO
>
> client = carla.Client("localhost", [...]

I am unable to find _GlobalRoutePlannerDAO_, if anyone have access to it please let me know.

Zhuweilong123, 2022-11-01:
@ManishwarG what is your CARLA version?

Saddy21, 2022-12-30:
@Zhuweilong123 mine is 0.9.13,there also i cant find GlobalRoutePlannerDAO.

TauJiang, 2023-01-03:
> 
an updated code for 0.9.13 is as below
```
import carla
import sys
sys.path.append('../')
from agents.navigation.global_route_planner import GlobalRoutePlanner
# from agents.navigation.global_route_planner_dao import GlobalRoutePlannerDAO

client = carla.Client("localhost", 2000)
client.set_timeout(10)
world = client.load_world('Town01')
amap = world.get_map()
sampling_resolution = 2
# dao = GlobalRoutePlannerDAO(amap, sampling_resolution)
grp = GlobalRoutePlanner(amap, sampling_resolution)
# grp.setup()
spawn_points = world.get_map().get_spawn_points()
print(spawn_points)
a = carla.Location(spawn_points[50].location)
b = carla.Location(spawn_points[100].location)
w1 = grp.trace_route(a, b) # there are other funcations can be used to generate a route in GlobalRoutePlanner.
i = 0
for w in w1:
    if i % 10 == 0:
        world.debug.draw_string(w[0].transform.location, 'O', draw_shadow=False,
        color=carla.Color(r=255, g=0, b=0), life_time=120.0,
        persistent_lines=True)
    else:
        world.debug.draw_string(w[0].transform.location, 'O', draw_shadow=False,
        color = carla.Color(r=0, g=0, b=255), life_time=1000.0,
        persistent_lines=True)
    i += 1
```